F5OEO/rpitx #103: Running setTime.py continuously

mrmattwilkins, 2018-04-12:
Hi,
I would like to use rpitx to run a DCF77 transmitter in my house.  I would like it to be running 24 hours a day, just going and going.  setTime.py appears to only run a certain number of minutes before quitting.  I know I could repeatedly start it up using cron, but is there a nicer solution?
Thanks
Matt

JsBergbau, 2018-04-13:
Call it with "-n" or "--nums" for number of minutes  of transmission

mrmattwilkins, 2018-04-13:
Hi,
I am aware of the -n, but I am wanting to run this forever, in other words
for an infinite number of minutes of transmission.  Is there a way of using
rpitx to output DCF77 continuously, just like a DCF77 transmitter?
Thanks
Matt

JsBergbau, 2018-04-13:
Then I suggest you edit line 159 in setTime.py
 `for i in range(1, NUMS + 1):`

to 

`while True:`

mrmattwilkins, 2018-04-14:
Hi,
Thanks for the suggestion, but I can't see how that will work.  Isn't that
loop building the file that will be later transmitted?   If I'm reading the
code correctly, a "while True" will result in an infinitely large file that
will never be transmitted?
Is the developer of rpitx on this forum?
Thanks
Matt

JsBergbau, 2018-04-14:
Sorry, you're right. The program first writes the file, with while True: infinetely large and thus never transmitted. But thats how the program works. Only looked too quickly, sorry.

>  I know I could repeatedly start it up using cron, but is there a nicer solution?
> Thanks

To need less work, try putting it into a bash script

```

#!/bin/sh

while true
do
./setTime.py -n XX
done
```
Thus it will be started again after XX minutes (replace how long you like running it continously)

closed 2018-11-23 by F5OEO