F5OEO/rpitx #106: RPITX : transmitting from TCP streaming

UgoPoddine, 2018-06-17:
Good morning,

I'm facing a problem in transmitting using RPITX from TCP streaming (generated by gnuradio).
The scenario is the following :

a) Raspbian Stretch 4.14.44+ #1117 on RaspberryPI2B+  armv6l GNU/Linux
b) rpitx Version 0.2 compiled Jun  4 2018
c) rpitx test script transmitting correctly on 10m : /rpitx -m IQ -i /home/pi/pat/ssbIQ.wav  -f 28150 -l
d) Gnuradio modulating the same sampleaudio.wav file in SSB / USB and sending it in complex format on TCP sink port (client mode). Waterfall shows modulated signal.
e) data streaming received by netcat and sent to rpitx as follow :

while true; do (nc -l 8011; dd if=/dev/zero bs=4096 count=30); done | sudo rpitx -i - -m IQ -f 28150

The result is that rpitx stays idle on the calibration phase and nothing is transmitted on air.

rpitx Version 0.2 compiled Jun  4 2018 (F5OEO Evariste) running on  Frequency=0.000000  SampleRate=48000 Master PLL = 1000000000
memory: 512 MB
processor: Broadcom BCM2835
i2cDevice: /dev/i2c-1
model: Model B+
manufacturer: Sony
pcb revision: 1
warranty void: no
revision: 0010
peripheral base: 0x20000000
Jessie
Using mbox device /dev/vcio.
3616000 Size NUM PAGES 883 PAGE_SIZE 4096
MASH 3 Freq PLL# 6
Calibrate : ppm=-10 DMA 160ns:1361ns

... and no more.

What I'm doing wrong ?
Any help appreciated
Thank-you
Best Regards, 73
Ugo Poddine IU1IPB

UgoPoddine, 2018-06-18:
Hello,

it could be correlated to the issue or not : I have remarked that a simple script like this : 

nc -u -l 8011 | sudo rpitx -i - -m IQ -f 28150

…receiving IQ stream from GnuRadio via UDP Sink seems to run correctly and transmitting on air if the data flow is issued short time after the startup of this script. If the gnuradio script starts later (more then 30 secs), RPITX doesn't start transmitting data and stays in "after calibration" status idle, as described above.
Is there some kind of timeout in RPITX data flow reception ? I have verified that it's not a problem of netcat -u (no timeout also after 5 minutes)…

Best regards
Ugo

F5OEO, 2018-06-18:
I am mainly working on v2beta branch. You coud try it OR if you still want to keep on the master branch : 
A new version (branch v2beta) will be released soon.
In the meantime you can try with master version adding  -c 1 -a 14 , output
RF is on GPIO4 pin 7

UgoPoddine, 2018-06-19:
Thank-you Evariste,, mainly for the great job you have done with rpitx.
I will try both : "c-1 -a 14" (what -a 14 does ?) and the v2beta.

Thank-you
Ugo

lu7did, 2018-11-01:
Hello, Is there any experience feeding rpitx with a PSK31 source? In the lines of piping the output of a program such as FLDIGI into rpitx for transmission. If not other experiences running PSK31 would be appreciated. Thanks, Pedro LU7DID

closed 2018-11-23 by F5OEO