F5OEO/rpitx #111: Bad quality, suggestions?

acujl, 2018-09-26:
Heya! I'm transmitting on 88Mhz but the audio is really bad quality, I have an antena (copper wire soldered) on the pin. Any suggestions?
Even without the homemade antena the audio is crap
I normaly do:
rpitx  -i /home/pi/SDRSharp_20180924_181735Z_AF.wav -f 88000

moliata, 2018-10-16:
First, try using a 10cm jumper wire instead and make sure, you are connecting it to a correct pin, that may help. Then try to increase the frequency to around 100MHz, a couple of MHz down or up. Last of all, try to use a different radio receiver, like a RTL-SDR (if you don't use it already).

acujl, 2018-10-26:
Thanks for the suggestions! Yes, I already use RTL-SDR 👍

F5OEO, 2018-11-23:
Fixed on version 2

closed 2018-11-23 by F5OEO