F5OEO/rpitx #112: PSK with gnuradio and rpitx

goulum29, 2018-10-24:
Hi, i'm trying to send a simple text file via PSK modulation by using gnuRadio and rpitx. I modulate with gnu radio and send via tcp socket to rpitx listening on tcp : 

` nc -l 9999 | sudo rpitx -i- -m IQFLOAT -f 67000`

The modulator : 

[image]

And i demodulate via gnu-radio by using rtl sdr.

[image]

I launch all these process and i revceive nothing,my ouptut file still at 0 bit , i can see on the spectrum and the waterfall that is a modulation but on the eye diagramm, i see that :

[image]

[image]

I try some different configuration about filtering and i get sometimes better constellation, but still transmit nothing. Sometimes, i have the "circle" of psk modulation but i clearly put in parameters 2 constellations point...
It's for personnal testing, i'm just trying to have a simple and functionnal digital modulation.

If i was wrong somewhere, if you have any idea ?  I take any advice, thank you in advance.

ha7ilm, 2018-10-24:
First could also try first putting together a simulation in a single flowgraph that contains both the `PSK Mod` and the `PSK Demod` blocks, and a channel model block between them, with some frequency offset and additive noise. That would allow you to make sure that the whole thing is likely to work with a pair of real SDRs.

Also make sure that the amplitude of the I/Q signal you're sending into RPiTX is correct. 

As a note, for PSK and FSK I'd rather use the `-m RF` mode instead, as without PWM it's likely that you'll get a cleaner signal. For PSK that requires some preprocessing though, so you won't be able to use the `PSK Mod` block then.

goulum29, 2018-10-24:
Ok , I will try that, but wich amplitude it's correct ? Amplitude it's not set by the carrier ?

ha7ilm, 2018-10-24:
Maybe it's correct if the signal uses the range between `+1.0` and `-1.0` (for both I and Q), but check the `rpitx` code / documentation.

F5OEO, 2018-10-24:
Be sure to set you samplerate to 48000 which is the default (from my memory). However, this branch is end of life and the new v2beta branch will be released soon. It is based on librpitx which has already some easy class to make FSK and PSK natively.

goulum29, 2018-10-24:
Ok, i will try all your advice and send you my return. And try FSK (gaussian) modulation.

goulum29, 2018-10-24:
I have test your proposal ha7ilm with channelizer, full software mod-demod with adding noise. It's work but when the noise rise up the signal seems pretty clear but 0 bit passed.
However i'm sorry but i don't understand wich parameters i need to pass to my flow graph and rpitx....
For me the IQFLOAT parameters accepts the complex output of gnuRadio. And i check my amplitude after modulation it's -1 and 1. 
I try with a 48k sample rate on my rtl-sdr source, i try band pass filter, but still not clear.
All my mod-demod works on software, with adding some noise but when passing to rpitx it's doesn't work.... I think i miss something

Thanks in advance

F5OEO, 2018-11-23:
For PSK , prefere to use MPSK modulator in V2 instead of using i/Q.

closed 2018-11-23 by F5OEO