F5OEO/rpitx #113: Error while installing v2beta branch

goulum29, 2018-10-26:
There is some installation error in the installation of rhe v2 beta branch : 
`imagemagick is already the newest version (8:6.9.7.4+dfsg-11+deb9u6).
0 upgraded, 0 newly installed, 0 to remove and 7 not upgraded.
gcc  -Wall -O3 -Wno-unused-variable  -c -o mailbox.o mailbox.c
gcc  -Wall -O3 -Wno-unused-variable  -c -o raspberry_pi_revision.o raspberry_pi_revision.c
g++  -Wall -O3 -Wno-unused-variable  -c dsp.cpp iqdmasync.cpp ngfmdmasync.cpp fmdmasync.cpp  dma.cpp gpio.cpp serialdmasync.cpp phasedmasync.cpp amdmasync.h amdmasync.cpp fskburst.cpp
ar rc librpitx.a dsp.o iqdmasync.o ngfmdmasync.o fmdmasync.o  dma.o gpio.o mailbox.o raspberry_pi_revision.o  serialdmasync.o phasedmasync.o amdmasync.o fskburst.o
g++ -Wall -g -O2 -Wno-unused-variable  -o ../sendiq sendiq.cpp  librpitx/src/librpitx.a -lm -lrt -lpthread  
gcc -Wall -g -O2 -Wno-unused-variable  -c -o dvb/dvbsenco8.o dvb/dvbsenco8.s
gcc -Wall -g -O2 -Wno-unused-variable  -c -o dvb/dvbs2arm_1v30.o dvb/dvbs2arm_1v30.s
gcc -Wall -g -O2 -Wno-unused-variable  -c -o dvb/fec100.o dvb/fec100.c
g++ -Wall -g -O2 -Wno-unused-variable  -o ../dvbrf dvb/dvbrf.cpp dvb/dvbsenco8.o dvb/fec100.o dvb/dvbs2arm_1v30.o librpitx/src/librpitx.a -lm -lrt -lpthread  	
install -m 0755 ../pisstv /usr/bin
install -m 0755 ../rpitx /usr/bin
Makefile:66: recipe for target 'install' failed
Installation done`

rpitx is not compilating and a few other like am or fm ?
Is it normal/in progress ?Or the errors can be resolve by compilating manually ?

F5OEO, 2018-10-26:
Should be fixed with https://github.com/F5OEO/rpitx/commit/260761ae1b2253ef6c5e104d2b579c134758e4ca

goulum29, 2018-10-26:
Apparently it's not : 
`install -m 0755 ../pisstv /usr/bin
install -m 0755 ../pifm /usr/bin
install: cannot stat '../pifm': No such file or directory
Makefile:66: recipe for target 'install' failed
make: *** [install] Error 1
Installation done
`

same error with copy paste the changes

F5OEO, 2018-10-26:
You have a mistake with your pulling process...Easiest one is to restart from sratch and cloning it. Works here.

goulum29, 2018-10-26:
Choosing the branch by using `git` and execute the script. It's working, thanks to you

closed 2018-10-26 by F5OEO