F5OEO/rpitx #118: Building error

TurtleForGaming, 2018-11-11:
So I was trying to build the latest version of rpitx unfortunatly I ran into an compilation error after some diggin it seems that the `abs()` function isn't defined in dsp.h and amdmasync.h. TO solve I've just included `#include <stdlib.h>` in these file and it compiled librpitx

closed 2018-11-11 by TurtleForGaming

F5OEO, 2018-11-11:
What your gcc version an kernel version?

TurtleForGaming, 2018-11-11:
`pi@raspberrypi:~/rpitx/src/librpitx/src $ gcc -v
Using built-in specs.
COLLECT_GCC=gcc
COLLECT_LTO_WRAPPER=/usr/lib/gcc/arm-linux-gnueabihf/4.9/lto-wrapper
Target: arm-linux-gnueabihf
Configured with: ../src/configure -v --with-pkgversion='Raspbian 4.9.2-10+deb8u1' --with-bugurl=file:///usr/share/doc/gcc-4.9/README.Bugs --enable-languages=c,c++,java,go,d,fortran,objc,obj-c++ --prefix=/usr --program-suffix=-4.9 --enable-shared --enable-linker-build-id --libexecdir=/usr/lib --without-included-gettext --enable-threads=posix --with-gxx-include-dir=/usr/include/c++/4.9 --libdir=/usr/lib --enable-nls --with-sysroot=/ --enable-clocale=gnu --enable-libstdcxx-debug --enable-libstdcxx-time=yes --enable-gnu-unique-object --disable-libitm --disable-libquadmath --enable-plugin --with-system-zlib --disable-browser-plugin --enable-java-awt=gtk --enable-gtk-cairo --with-java-home=/usr/lib/jvm/java-1.5.0-gcj-4.9-armhf/jre --enable-java-home --with-jvm-root-dir=/usr/lib/jvm/java-1.5.0-gcj-4.9-armhf --with-jvm-jar-dir=/usr/lib/jvm-exports/java-1.5.0-gcj-4.9-armhf --with-arch-directory=arm --with-ecj-jar=/usr/share/java/eclipse-ecj.jar --enable-objc-gc --enable-multiarch --disable-sjlj-exceptions --with-arch=armv6 --with-fpu=vfp --with-float=hard --enable-checking=release --build=arm-linux-gnueabihf --host=arm-linux-gnueabihf --target=arm-linux-gnueabihf
Thread model: posix
gcc version 4.9.2 (Raspbian 4.9.2-10+deb8u1)
pi@raspberrypi:~/rpitx/src/librpitx/src $ uname -a
Linux raspberrypi 4.9.35-v7+ #1014 SMP Fri Jun 30 14:47:43 BST 2017 armv7l GNU/Linux
`

F5OEO, 2018-11-11:
Sorry , I assumed a latest stretch version ; here 
gcc version 6.3.0 20170516 (Raspbian 6.3.0-18+rpi1+deb9u1) 
Linux PI2Stretch1 4.14.78-v7+

TurtleForGaming, 2018-11-11:
You should put a check in the install.sh script so user don't get confused because a after this error this is an error with pifmrds. And thanks for the solution