F5OEO/rpitx #119: How to transmit AM OOK?

extinctpotato, 2018-11-12:
I have a 315 MHz remote. I recorded it using GQRX and processed in Audacity so that it is 48kHz Mono 16-bit signed WAV. Now I have absolutely no idea how to send it. The 'piam' utility seems to be deprecated and in general, while your project looks very interesting, there is little to no documentation available, sadly.

When I fire the 'testam.sh' script it just sends a carrier wave. I believe it is a bug.

My hardware is Raspberry Pi B.

Thank you for your help sincerely!

F5OEO, 2018-11-12:
See the rtlmenu : record and replay : 
Record your remote :
rtl_sdr -s 250000 -g 30 -f 315M record.iq 
Replay it with : 
 sudo ./sendiq -s 250000 -f 315e6 -t u8 -i record.iq 

That's it !

extinctpotato, 2018-11-12:
It works! Sadly, it introduces another problem. When it transmits, the Ethernet disconnects. Maybe it should use different pin?

F5OEO, 2018-11-12:
Nice . As per readme, Pi B is partially supported. Need to investigate (not priority) or you can invest in a low cost Pizero !

extinctpotato, 2018-11-12:
Oh, okay. So that's probably why.
Btw, I tried the V1. When I use it, it does send the data. It is distorted (therefore doesn't trigger the remote device) but Ethernet does not disconnect. I believe it uses a different pin or mode.
Is there any way to make 'sendiq' use a different pin? If there is I think it could work!

F5OEO, 2018-11-24:
Can you try with latest v2 version. There was a conflict with Wifi but could be the same issue in PIB with ethernet.

dachshund-digital, 2019-04-04:
Did some testing, and I don't see Ethernet disconnecting in my setup.  I setup a constant ping cycle on the Pi in an SSH session.  Then in a different ssh session, ran several sendiq commands.... turning on and off light on a ceiling fan, using 303.875 MHz.  My antenna is on GPIO 4, is a 20 CM jumper wire.  The Pi I am using for this test is about 4 meters from the ceiling fan.  Using the latest stable rpitx release pulled it from here just this morning.