F5OEO/rpitx #120: Error / warning when compiling v2

pa7r, 2018-11-17:
When  try compile V2 on Raspbian version 8 (Jessie) with gcc updated to 4.9.2 
The following (severe) warnings appear;

In file included from dsp.cpp:19:0:
dsp.h:12:22: warning: non-static data member initializers only available with -std=c++11 or -std=gnu++11
  double prev_phase = 0;
dsp.h:12:22: warning: non-static data member initializers only available with -std=c++11 or -std=gnu++11
  double prev_phase = 0;
                      ^

etc etc

Any ideas?

73`s,

Rene, PA7R

F5OEO, 2018-11-17:
As per readme, I assume a stretch version of Raspbian. See my gcc version : 
gcc version 6.3.0 20170516 (Raspbian 6.3.0-18+rpi1+deb9u1) .
I will investigate for Jessie, but simpler is to upgrade you system.

pa7r, 2018-11-17:
Thanks for the hint. I upgraded he Raspberry to Stretch ( gcc is now 6.3.0 20170516)
and installed again , no errors this time!
The rpitxV2 seems to  be working fine in all modes with easytest. I just cant stop the loop with CTRL +C
The Raspberry V2 has no screen or keyboard attached to it, just using Putty or VNC
Even a second session in Putty does not give any respond when the Raspberry is transmitting
Only power off / Power on seems to do the trick

73`s,

Rene, PA7R

closed 2018-11-17 by F5OEO