F5OEO/rpitx #122: Easytest not stopping

F5OEO, 2018-11-17:
and installed again , no errors this time!
The rpitxV2 seems to  be working fine in all modes with easytest. I just cant stop the loop with CTRL +C
The Raspberry V2 has no screen or keyboard attached to it, just using Putty or VNC
Even a second session in Putty does not give any respond when the Raspberry is transmitting
Only power off / Power on seems to do the trick

73`s,

Rene, PA7R

_Originally posted by @pa7r in https://github.com/F5OEO/rpitx/issues/120#issuecomment-439623922_

F5OEO, 2018-11-17:
@pa7r If you start sstv in easytest and let it finish, does it return to menu ? Which Pi Model do you use ?

pa7r, 2018-11-17:
OK,
Model is a Raspberry 2B, yes I use SSH (Putty) over Wifi
When the SSTV is finished it does not return to the easytest menu, the wifi connection seems to be aborted after this

F5OEO, 2018-11-17:
This is caused by Wifi : related to https://github.com/F5OEO/rpitx/issues/116 . Need to investigate. If you have any chance to link it by ethernet, confirm that all is working.

pa7r, 2018-11-18:
Yes,

 if RpitxV2 is transmitting, and the Ethernet interface is physically connected, the Wifi interface stays alive
All is working, the Rpi responds to a CTRL +C
It makes no difference if static adress are being used or not. When the Ethernet cable is detached, wifi stops working

F5OEO, 2018-11-18:
OK, thanks for report. If you are on Rpi2B you surely use a Wifi usb stick, which is a bit strange as I guess wifi issue related is with integrated raspberry devices. Need to reproduce and investigate , stay tuned ;)

pa7r, 2018-11-18:
Fortunately I have a Rpi3 with integrated wifi,, but this one has to be upgraded to Stretch as well.
Ill post my findings when the upgrade is completed

F5OEO, 2018-11-23:
Should be fixed with https://github.com/F5OEO/librpitx/commit/6148190afa00ef47ccfbf7d04ac5b28ec529ce87
To update :
git pull librpitx and compile
compile rpitx and install

Let me know if it works.

pa7r, 2018-11-23:
I installed everything from scratch, rpitx is working fine now, no wifi hangup anymore

73`s

Rene, PA7R

Should be fixed with F5OEO/librpitx@6148190<https://github.com/F5OEO/librpitx/commit/6148190afa00ef47ccfbf7d04ac5b28ec529ce87>

F5OEO, 2018-11-23:
Thanks for raising up the issue. Not always easy to test all configurations.

closed 2018-11-23 by F5OEO