F5OEO/rpitx #125: AF shift issue.

maxlock, 2018-11-26:
I'm generating a 1khz tone using sox
```
sox -V -r 48000 -n -b 16 -c 2 note.wav synth 15 sin 1000
```
then converting for rpitx
```
cat note.wav | csdr convert_i16_f | csdr gain_ff 7000 | csdr convert_f_samplerf 20833 > note_processed.wav
```
and finally transmitting 
```
sudo rpitx -i ./note_processed.wav -m RF -f 434000
```

The received tone is at 500hz, with additional harmonics at 1000,1500 & 200hz etc. I noticed this because my generated aprs packets sounded off, and aren't being decoded properly. Is there anything that could cause a tone shift happening? if I play back the 1khz wav locally it's fine, and there are no harmonics either.

I've also sent the sampleaudio.wav and it sounds ok to me. is there additionally something I could change in the csdr conversion?

I'm using a USB attached original pi zero and version 2.0

Thanks, Max. G7UOZ.

F5OEO, 2018-11-27:
Interesting question : first , you should use a mono file : 
`sox -V -r 48000 -n -b 16 -c 1 note.wav synth 15 sin 1000`

As per https://en.wikipedia.org/wiki/Frequency_modulation, see Bessels functions . Here modulation deviation is 7khz and modulation frequency is 1khz, means a modulation index of 7. Spectrum seems to correspond to theory ?

maxlock, 2018-11-28:
Interesting indeed, when running with a single channel wav, everything is fine. I'll keep digging and thanks for the bessels function link, good reading :)

closed 2018-11-28 by maxlock

maxlock, 2018-11-28:
ok, so it seems that the source wav must have a 48khz rate, the following will generate two 1khz tones but will  be pitch shifted when transmitted.

```
sox -V -r 44000 -n -b 16 -c 1 note44.wav synth 15 sin 1000
sox -V -r 48000 -n -b 16 -c 1 note48.wav synth 15 sin 1000

cat note44.wav | csdr convert_i16_f | csdr gain_ff 7000 | csdr convert_f_samplerf 20833 > note44tx.wav
cat note48.wav | csdr convert_i16_f | csdr gain_ff 7000 | csdr convert_f_samplerf 20833 > note48tx.wav

sudo rpitx -i ./note44tx.wav -m RF -f 434000
sudo rpitx -i ./note48tx.wav -m RF -f 434000
```

F5OEO, 2018-11-28:
You can set the symbolrate with rpitx : -s option (sorry , not in the help of command line)
By default it is 48Khz.

maxlock, 2018-11-28:
Aha! excellent, thanks again! :)

F5OEO, 2018-11-28:
Note that csdr convert_f with parameter 20833=1/48000 is not used anymore with version 2.

maxlock, 2018-11-30:
OK, I was using the test_nfm.sh script (below) as a guide. what would be your suggestion for nbfm then?
```
(while true; do cat sampleaudio.wav; done) | csdr convert_i16_f | csdr gain_ff 7000 | csdr convert_f_samplerf 20833 | sudo ./rpitx -i- -m RF -f $1
```