F5OEO/rpitx #136: POCSAG at 512 bps don't work

pyrog, 2018-12-10:
In [pocsag.cpp#L332](https://github.com/F5OEO/rpitx/blob/master/src/pocsag/pocsag.cpp#L332):
```cpp
	int SR=1200;
```

With `SR=1200` or `SR=2400`, messages are decoded with [multimon-ng](https://github.com/EliasOenal/multimon-ng).
But unfortunately, not with `SR=512` (tested with multimon-ng and a real pager).

Could encode a pocsag message with [faithanalog/pocsag-encoder](https://github.com/faithanalog/pocsag-encoder) and decode it with multimon-ng:

```bash
printf "0123456:Test message" \
 | pocsag-encoder/pocsag \
 | multimon-ng -a POCSAG512 - -q -f alpha
```

I tried to generate POCSAG IQ samples and send them to `sendiq`…
But I haven't enough knowledge to convert the audio pcm output of `pocsag-encoder`:
```bash
printf "0123456:Test message" \
| pocsag-encoder/pocsag \
| ppm to IQ conversion \
| sudo sendiq -s 22050 -i /dev/stdin -f 439.9875e6
```

F5OEO, 2018-12-14:
Thx for your experiments. I need to double check with multimon-ng and real pager to fix issues. You can also contribute !

pyrog, 2018-12-17:
> You can also contribute !

Bien sûr 😊

Je peux ajouter des paramètres pour choisir la vitesse de transmission,  les bits de fonction, le format des messages (numérique, alphanumérique) et inverser si besoin le signal (cf. #135).

Par contre, je ne sais pas comment vérifier si le signal émis en 512 bauds est valide…

pyrog, 2018-12-27:
🤭pocsag transmission work with `SR=512` ! 

I probably forgot to answer `y` at last question of `install.sh`
`In order to run properly, rpitx need to modify /boot/config.txt. Are you sure (y/n)`

closed 2018-12-27 by pyrog

F5OEO, 2018-12-27:
Oh good news !

pyrog, 2018-12-27:
I just test it with a real pager (Swissphone DE700):
it beep at 3 meters without any antenna on the Raspberry pi ! 😀

Povezi, 2018-12-31:
Is /boot/config.txt modification necessary for Rpi3B+ (with raspbian) ?

pyrog, 2019-01-01:
I guess yes.

Did you try the -i option ?

Povezi, 2019-01-01:
Yes, I have tried adding -i (invert) but only using
 sudo ./pocsag -f "160600000" -r 512
without -i (changes icon on pager - it detect signal - so I suppose baudrate, freqency and no inversion are ok), I see pager number on it, but if I try:
printf "1:0705967\n2:Helloworld" | sudo ./pocsag -f "160600000" -r 512
it doesn't receive it.
I have tried also:
printf "0705967:Helloworld" | sudo ./pocsag -f "160600000" -r 512
but the same result (detects signal, doesn't receive message)

any ideas for address:message format?

pyrog, 2019-01-01:
> any ideas for address:message format?

See https://github.com/F5OEO/rpitx/issues/121#issuecomment-450761038

To ring your pager, you must use the good frequency, baud rate, "polarity" and address.