F5OEO/rpitx #139: rpitx stucks on running

BRRY876, 2018-12-22:
after installing the rpitx on raspberry pi 3b+ on raspbian using bash install.sh command, i tried to use various tools such as easytest.sh..i tried chirp and all other tools and it stucks on  a window transmitting..based on the pics for the tool another window should have opened with showing the frequency graph

specs on my device:
Linux raspberrypi 4.14.79-v7+ #1159 SMP Sun Nov 4 17:50:20 GMT 2018 armv7l GNU/Linux

F5OEO, 2018-12-22:
rpitx seems working. You should read more carefully the Readme to understand what you are running. Graphical software is a third party SDR GUI (Sdrconsole for example) running on an other PC as a receptor.

BRRY876, 2018-12-22:
its working fine... had a misunderstanding

closed 2018-12-22 by BRRY876