F5OEO/rpitx #142: Can not make file

Commander-Crash, 2018-12-25:
I have been trying to make rpitx. All is good with scripts, rtl menu and test menu, However pocsag and pichirp will not make. I may be missing something but can not see what. I used ./install.sh with chmod +x perimeter.

Raspberry pi 3B info::
Distributor ID: Raspbian
Description:    Raspbian GNU/Linux 8.0 (jessie)
Release:          8.0
Codename:     jessie

root@raspberrypi:/home/pi/rpitx# ./install.sh
Install rpitx - some package need internet connection -
Hit:1 http://downloads.metasploit.com/data/releases/metasploit-framework/apt lucid InRelease
Hit:2 http://archive.raspberrypi.org/debian jessie InRelease
Hit:3 http://mirrordirector.raspbian.org/raspbian jessie InRelease
Ign:4 http://giteduberger.fr rpimonitor/ InRelease
Get:5 http://giteduberger.fr rpimonitor/ Release [1,396 B]
Fetched 1,396 B in 5s (249 B/s)
Reading package lists... Done
Reading package lists... Done
Building dependency tree
Reading state information... Done
git is already the newest version (1:2.1.4-2.1+deb8u7).
libsndfile1-dev is already the newest version (1.0.25-9.1+deb8u1).
0 upgraded, 0 newly installed, 0 to remove and 387 not upgraded.
Reading package lists... Done
Building dependency tree
Reading state information... Done
imagemagick is already the newest version (8:6.8.9.9-5+deb8u15).
libfftw3-dev is already the newest version (3.3.4-2+rvt).
0 upgraded, 0 newly installed, 0 to remove and 387 not upgraded.
Reading package lists... Done
Building dependency tree
Reading state information... Done
buffer is already the newest version (1.19-12).
rtl-sdr is already the newest version (0.5.3-3).
0 upgraded, 0 newly installed, 0 to remove and 387 not upgraded.
fatal: destination path 'csdr' already exists and is not an empty directory.
patching file csdr/Makefile
Reversed (or previously applied) patch detected!  Assume -R? [n] y
install -m 0755 libcsdr.so.0.15 /usr/lib
install -m 0755 csdr /usr/bin
install -m 0755 csdr-fm /usr/bin
install -m 0755 nmux /usr/bin
#-install -m 0755 ddcd /usr/bin
fatal: destination path 'librpitx' already exists and is not an empty directory.
cc -Wall -O3 -Wno-unused-variable -c -o mailbox.o mailbox.c
cc -Wall -O3 -Wno-unused-variable -c -o raspberry_pi_revision.o raspberry_pi_revision.c
c++ -std=c++11 -Wall -O3 -Wno-unused-variable -c dsp.cpp iqdmasync.cpp ngfmdmasync.cpp fmdmasync.cpp  dma.cpp gpio.cpp serialdmasync.cpp phasedmasync.cpp amdmasync.h amdmasync.cpp fskburst.cpp
ar rc librpitx.a dsp.o iqdmasync.o ngfmdmasync.o fmdmasync.o  dma.o gpio.o mailbox.o raspberry_pi_revision.o  serialdmasync.o phasedmasync.o amdmasync.o fskburst.o
g++ -Wall -g -O2 -Wno-unused-variable  -o ../pisstv sstv/pisstv.cpp  librpitx/src/librpitx.a -lm -lrt -lpthread
In file included from sstv/../librpitx/src/dma.h:4:0,
                 from sstv/../librpitx/src/librpitx.h:18,
                 from sstv/pisstv.cpp:18:
sstv/../librpitx/src/gpio.h:12:32: warning: non-static data member initializers only available with -std=c++11 or -std=gnu++11 [enabled by default]
sstv/../librpitx/src/gpio.h:256:29: warning: non-static data member initializers only available with -std=c++11 or -std=gnu++11 [enabled by default]
sstv/../librpitx/src/gpio.h:257:28: warning: non-static data member initializers only available with -std=c++11 or -std=gnu++11 [enabled by default]
sstv/../librpitx/src/gpio.h:259:17: warning: non-static data member initializers only available with -std=c++11 or -std=gnu++11 [enabled by default]
sstv/../librpitx/src/gpio.h:261:20: warning: non-static data member initializers only available with -std=c++11 or -std=gnu++11 [enabled by default]
sstv/../librpitx/src/gpio.h:329:29: warning: non-static data member initializers only available with -std=c++11 or -std=gnu++11 [enabled by default]
sstv/../librpitx/src/gpio.h:330:14: warning: non-static data member initializers only available with -std=c++11 or -std=gnu++11 [enabled by default]
In file included from sstv/../librpitx/src/librpitx.h:18:0,
                 from sstv/pisstv.cpp:18:
sstv/../librpitx/src/dma.h:68:18: warning: non-static data member initializers only available with -std=c++11 or -std=gnu++11 [enabled by default]
sstv/../librpitx/src/dma.h:81:15: warning: non-static data member initializers only available with -std=c++11 or -std=gnu++11 [enabled by default]
In file included from sstv/../librpitx/src/iqdmasync.h:7:0,
                 from sstv/../librpitx/src/librpitx.h:22,
                 from sstv/pisstv.cpp:18:
sstv/../librpitx/src/dsp.h:12:22: warning: non-static data member initializers only available with -std=c++11 or -std=gnu++11 [enabled by default]
In file included from sstv/../librpitx/src/librpitx.h:22:0,
                 from sstv/pisstv.cpp:18:
sstv/../librpitx/src/iqdmasync.h:18:13: warning: non-static data member initializers only available with -std=c++11 or -std=gnu++11 [enabled by default]
[... 91 lines cut ...]
chirp/../librpitx/src/gpio.h:261:20: warning: non-static data member initializers only available with -std=c++11 or -std=gnu++11 [enabled by default]
chirp/../librpitx/src/gpio.h:329:29: warning: non-static data member initializers only available with -std=c++11 or -std=gnu++11 [enabled by default]
chirp/../librpitx/src/gpio.h:330:14: warning: non-static data member initializers only available with -std=c++11 or -std=gnu++11 [enabled by default]
In file included from chirp/../librpitx/src/librpitx.h:18:0,
                 from chirp/chirp.cpp:2:
chirp/../librpitx/src/dma.h:68:18: warning: non-static data member initializers only available with -std=c++11 or -std=gnu++11 [enabled by default]
chirp/../librpitx/src/dma.h:81:15: warning: non-static data member initializers only available with -std=c++11 or -std=gnu++11 [enabled by default]
In file included from chirp/../librpitx/src/iqdmasync.h:7:0,
                 from chirp/../librpitx/src/librpitx.h:22,
                 from chirp/chirp.cpp:2:
chirp/../librpitx/src/dsp.h:12:22: warning: non-static data member initializers only available with -std=c++11 or -std=gnu++11 [enabled by default]
In file included from chirp/../librpitx/src/librpitx.h:22:0,
                 from chirp/chirp.cpp:2:
chirp/../librpitx/src/iqdmasync.h:18:13: warning: non-static data member initializers only available with -std=c++11 or -std=gnu++11 [enabled by default]
In file included from chirp/../librpitx/src/librpitx.h:24:0,
                 from chirp/chirp.cpp:2:
chirp/../librpitx/src/phasedmasync.h:12:16: warning: non-static data member initializers only available with -std=c++11 or -std=gnu++11 [enabled by default]
chirp/chirp.cpp: In function ‘int main(int, char**)’:
chirp/chirp.cpp:58:90: error: ‘exit’ was not declared in this scope
chirp/chirp.cpp:60:30: error: ‘atof’ was not declared in this scope
Makefile:26: recipe for target '../pichirp' failed
make: *** [../pichirp] Error 1
\n\n
In order to run properly, rpitx need to modify /boot/config.txt. Are you sure (y/n) y
Warning : Rpitx should be instable and stop from transmitting !

F5OEO, 2018-12-25:
You seem to run "Codename: jessie". I assume "stretch" to be as a minimal to avoid backports and maintenance. Please upgrade or provide patches for your version.

Commander-Crash, 2018-12-28:
Ok thank you I have stretch on a different pi. It was the issue with jessie.

closed 2018-12-28 by Commander-Crash