F5OEO/rpitx #143: Can't install and compile, shows me errors

All3xJ, 2018-12-25:
I am using kali-pi.

Linux kali-pi 4.14.62-Re4son-v7+ #2 SMP Sat Sep 22 22:53:40 AEST 2018 armv7l GNU/Linux

```
mailbox.c:(.text+0x448): undefined reference to `makedev'
/usr/bin/ld: mailbox.c:(.text+0x482): undefined reference to `makedev'
collect2: error: ld returned 1 exit status
make: *** [Makefile:16: ../pisstv] Error 1
c++ -std=c++11 -Wall -g -O2 -Wno-unused-variable -o ../pisstv sstv/pisstv.cpp  librpitx/src/librpitx.a -lm -lrt -lpthread
/usr/bin/ld: librpitx/src/librpitx.a(mailbox.o): in function `mbox_open':
mailbox.c:(.text+0x448): undefined reference to `makedev'
/usr/bin/ld: mailbox.c:(.text+0x482): undefined reference to `makedev'
collect2: error: ld returned 1 exit status
make: *** [Makefile:16: ../pisstv] Error 1
```

F5OEO, 2018-12-26:
Can you try to install makedev prior to compile : sudo apt-get install makedev

All3xJ, 2018-12-26:
> Can you try to install makedev prior to compile : sudo apt-get install makedev

hi, thanks for reply. I did it, but the same errors show.

All3xJ, 2018-12-26:
I solved by adding` #include <sys/sysmacros.h>` in file `rpitx/src/librpitx/src/mailbox.c`

F5OEO, 2019-01-06:
Besides this, have you succeed in compiling the entire project and run it ?

jemussi, 2020-03-07:
> 
> 
> I solved by adding` #include <sys/sysmacros.h>` in file `rpitx/src/librpitx/src/mailbox.c`

Thanks, this worked for me too. Had the same issue while to trying run make when compilling WsppryPi but appears to work fine now.