F5OEO/rpitx #145: NFM, AM, SSB not working in easytest

Toni43, 2018-12-28:
Hello.
NFM, AM, and SSB modulation do not work in easytest.sh  When I start any of this modulations, only a very short carrier is transmitted to the air, and then silence.
I use:
Raspberry B+ v1.2, Linux raspberrypi 4.14.79+ #1159 Sun Nov 4 17:28:08 GMT 2018 armv6l GNU/Linux

I tried to do this:
_(while true; do cat sampleaudio.wav; done) | csdr convert_i16_f   | csdr fir_interpolate_cc 2 | csdr dsb_fc   | csdr bandpass_fir_fft_cc 0.002 0.06 0.01 | csdr fastagc_ff   | sudo ./sendiq -i /dev/stdin -s 96000 -f 144500000 -t float_
But it works the same way

Output is - 

_pi@raspberrypi:~/rpitx $ (while true; do cat sampleaudio.wav; done) | csdr convert_i16_f   | csdr fir_interpolate_cc 2 | csdr dsb_fc   | csdr bandpass_fir_fft_cc 0.002 0.06 0.01 | csdr fastagc_ff   | sudo ./sendiq -i /dev/stdin -s 96000 -f 144500000 -t float
Channel 14 CBSize 64000 UsermemSize 48000
Using mbox device /dev/vcio.
2240000 Size NUM PAGES 547 PAGE_SIZE 4096
BufferSize 16000 , cb 64000 user 48000
Found PLL solution for frequency 144.5MHz : divider:11 VCO: 1589.5MHz
Master PLLC Locked
PCM prediv = 10
PCM clk=520 / 3413
Loop parameter =1a0000
End of file_

F5OEO, 2018-12-28:
Do you have the line GPU_FREQ=250 in /boot/config.txt ?

Toni43, 2018-12-28:
No, there is no this line in  /boot/config.txt 
I have only this 
#core_freq=250

F5OEO, 2018-12-28:
Install.sh include : 
 echo "Set GPU to 250Mhz in order to be stable"
   LINE='gpu_freq=250'
   FILE='/boot/config.txt'

If you haven't used install.sh or answer no at the last question , you need to add 
grpu_freq=250 in /boot/config.txt

Toni43, 2018-12-28:
Ok, thanks. Now it work.

closed 2019-01-06 by F5OEO

steve2162, 2022-11-29:
Hi, I've got the same issues as toni43. It works ok on an RPI3, but I need to use that on Gqrx for receive  so I thought I'd use a spare PI2 I have. It gives exactly the same symptoms as toni43. When I try to use ./testssb.sh or ./testnfm.sh it repeats "Illegal instruction" four times then "Peri Base = 20000000 SDRAM 40000000". I get a single 0.5 second burst of carrier then nothing. Both PI's are on "Raspbian GNU/Linux 11 (bullseye)" wide band FM,rtty and more all work as expected. The /boot/config.txt does have an entry of "core_freq=250
sdram_freq=400
over_voltage=0
gpu_freq=250
force_turbo=1"
Can you shed any light on this pease ?
Regards Steve W. (G8FGB)