F5OEO/rpitx #151: pocsag broken

F5OEO, 2019-01-10:


pyrog, 2019-01-13:
Unfortunately, POCSAG is broken 😔

Tested with all the following values, **multimon-ng detect POCSAG messages**, but TPL and Swissphone **pagers don't**.

```C
float Deviation=4500; 
```
```C
float Deviation=4500*2; 
```
```C
float Deviation=4500/2; 
```

How to be sure that pocsag use the last version of librpitx ?

F5OEO, 2019-01-13:
Be sure to go rpitx/src/librpitx/src and make
and then make on /rpitx/src , sudo make install.

I need to make a debug message to know which version of librpitx we use.

F5OEO, 2019-01-13:
Seems to be fixed

closed 2019-01-13 by F5OEO

Povezi, 2019-01-14:
yes, pulled lib, make, then make rpitx. tried both without -i and with it. Hackrf detects briefly bar (frequency is good) but didn’t echo address:msg in pocsag receiver