F5OEO/rpitx #153: the install script doesn't create a new line

misaalanshori, 2019-01-12:
[image]
 and it did that, so i was confused why i couldn't connect via the usb cable....

F5OEO, 2019-01-12:
I didn't test with NOOBS (assume we are on Raspbian strecth based).
It seems that you should maybe set gpu_freq=250 in the config.txt which is not located on /boot in NOOBS.

dwc2gpu has no relation with rpitx.

dave3825us, 2019-01-12:
I am new to linux but got everything installed. But want to know on a pi 3b+, what exactly is rhe gpu and will the fact that this prog changed it, will it affect the pi performance with other programs? Thanks

misaalanshori, 2019-01-12:
Im not sure what the default value for the gpu_freq is, but if it's a big change it could affect other programs that use the gpu... But rpitx needs to change it to be stable

closed 2019-01-25 by misaalanshori