F5OEO/rpitx #157: SSB test only transmits properly on 5 KHz steps

2WR3505, 2019-01-29:
I was just testing out RPiTX with the different modes, i think the SSTV test might have the same issue.

it transmits ok on like 3.845 or 3.835 and ect., but if i try anything like 384 or 3.842, it's messy as hell.

is this a problem software side or a hardware issue?

F5OEO, 2019-01-29:
This is right now an known issue not documented : some frequencies are not well handled as it cross two integer steps of master PLL. Normally there is a warning " cross boundary frequency".
Example : 
```
sudo ./tune -f 384e6
Info:NTP find offset -272669 freq -664222 pps=0 ppm=-10.135223
Info:NTP find offset -272669 freq -664222 pps=0 ppm=-10.135223
Freq PLL no 5= 767999982
Warning : cross boundary frequency
Found PLL solution for frequency 384.0MHz : divider:2 VCO: 768.0MHz
```
This a special case which usually doesn't appear on HamRadio bands and ISM bands, but as however need to be fixed.

2WR3505, 2019-01-29:
ah, yea, i didnt try the tune test down at 3.840, it prolly would've gave me the same warning. 

thanks for the info and hopefully things will work out. im lookin forward to talkin to my neighbor on the warfa QSO channel.