F5OEO/rpitx #161: RPi 3 randomly stops transmitting

THECRUSTIEST, 2019-02-28:
For all modes, all scripts, _anything_ that uses rpitx, transmission almost always stops early. This occurs usually after 4 or 5 seconds. Unlike other issues, the carrier disappears as well. Below is an image of three attempted SSTV transmissions from the "easytest" menu (not looped). I'm running Stretch. Transmissions are very clear, no issues, but they just randomly cut off. I tried adding "force_turbo=1" to /boot/config.txt, but nothing changed. Pi works fine, everything else is stable. Can't figure out what's wrong here 😢 

[image]

F5OEO, 2019-02-28:
Do you have gpu_freq=250 in /boot/config.txt ?

THECRUSTIEST, 2019-03-01:
Thank you for such a quick reply! I do have that line in config.txt. The end of my config is as follows:
force_turbo=1
arm_freq=700
gpu_freq=250
core_freq=250
sdram_freq=400

THECRUSTIEST, 2019-03-01:
Did a fresh Raspbian install, wiped everything, and now it seems to work fine. Perhaps there was some sort of interfering process. Thanks!

closed 2019-03-01 by THECRUSTIEST

Ghaz75, 2020-07-17:
No luck for me at least. In RPi3 it transmits for 3 to 4 seconds for all the modulations using ./easytest.sh, but in RPi2 it works fine. Only SSB works for around 60 seconds.