F5OEO/rpitx #163: Add GPIO Tx number pin

F5OEO, 2019-03-04:


dnhuan, 2019-08-16:
@F5OEO  Have you implemented this?

F5OEO, 2019-08-16:
Not yet on all command lines

dnhuan, 2019-08-16:
My GPIO4 is currently busy, is there anyway to change GPIO? I read your answer here but don't know exactly where to edit the GPIO.

> https://groups.io/g/rpitx/message/32

jbdu94, 2020-05-19:
Hi !
same issue here, my gpio 4 is busy with LCD display
How can I use rpitx with pin 32 (aka gpio 12) ?
Please provide instructions how to change this, I am sure that many users would find it interesting ;)
thx for reading, hope u can help

dnhuan, 2021-09-05:
https://github.com/F5OEO/rpitx/issues/62#issuecomment-302081448

Eham123, 2025-09-22:
Same problem with mine, GPIO4 is being used by a display and I need a way to change it to GPIO5 which is barely out of the way of my screen.