F5OEO/rpitx #164: POCSAG/FMRDS interfering with mplayer

gr01d, 2019-03-09:
I installed this on my Pi 3B+, I tested the POCSAG and FMRDS programs.
After that, I played a music stream on mplayer but it says:

MPlayer 1.3.0 (Debian), built with gcc-6.2.1 (C) 2000-2016 MPlayer Team
do_connect: could not connect to socket
connect: No such file or directory
Failed to open LIRC support. You will not be able to use your remote control.

Playing http://sg-icecast.eradioportal.com:8000/febc_dzfe.
Resolving sg-icecast.eradioportal.com for AF_INET...
Connecting to server sg-icecast.eradioportal.com[202.55.90.209]: 8000...

Website: icecast.eradioportal.com:8000/febc_dzfe
Public : yes
Bitrate: 128kbit/s
Cache size set to 320 KBytes
Cache fill:  0.00% (0 bytes)   
ICY Info: StreamTitle='';

Audio only file format detected.
==========================================================================
Opening audio decoder: [mpg123] MPEG 1.0/2.0/2.5 layers I, II, III
AUDIO: 44100 Hz, 2 ch, s16le, 128.0 kbit/9.07% (ratio: 16000->176400)
Selected audio codec: [mpg123] afm: mpg123 (MPEG 1.0/2.0/2.5 layers I, II, III)
==========================================================================
Failed to stat runtime directory /home/pi/.config/pulse/5e3a8d5cbaff427898fd4b9ec446e38e-runtime: Invalid argument
AO: [pulse] Init failed: Connection refused
Failed to initialize audio driver 'pulse'
AO: [alsa] 44100Hz 2ch s16le (2 bytes per sample)
Video: no video
Starting playback...
A:  -0.0 (unknown) of -0.0 (unknown) ??,?% 16% 
Audio device got stuck!
A:  -0.0 (unknown) of -0.0 (unknown) ??,?% 16% 
Audio device got stuck!
A:  -0.0 (unknown) of -0.0 (unknown) ??,?% 16% 
Audio device got stuck!
A:  -0.0 (unknown) of -0.0 (unknown) ??,?% 16% 
....
Ctrl+C

Until I reboot the Pi and do not use pocsag/fmrds, mplayer works.

gr01d, 2019-03-09:
I apologize to this post, I cannot edit and 'insert code'. Browser issue.

SarahRoseLives, 2024-07-19:
I have a similar issue on the same model pi where once I transmit with rpitx my audio doesn't work anymore. I can't use aplay or anything, it fails to connect or play every time.

It also appears to lock up various software until reboot which uses audio input/output