F5OEO/rpitx #166: Installed latest rpitx and piam execute does not exist?

dachshund-digital, 2019-04-03:
Installed latest rpitx and piam execute does not exist?

Searched entire file system and find no binary or script file named 'piam' anywhere.  Only found reference to a source code file... piam.c.

Searched all packages available via apt-file command no piam referenced by any package, but clearly it is expected to be part of the rpitx software no?

tbird635, 2019-05-30:
I have the same problem. Can't convert wav files because there's no PIAM file included.

psa-jforestier, 2020-09-01:
It seems the PIAM build step in the Makefile has been removed.
Add the following lines in the makefile in src, just before `../rpitx' : 
```
CFLAGS_Piam   = -Wall -g -O2 -Wno-unused-variable
LDFLAGS_Piam  = librpitx/src/librpitx.a -lm -lrt -lpthread -lsndfile -L/opt/vc/lib -lbcm_host
../piam: librpitx/src/librpitx.a
        $(CC) $(CFLAGS_Piam) -o ../piam am/piam.c  $(LDFLAGS_Piam)
```
/!\ Line with $(CC) start with a TAB.

Then, enter `make ../piam` to build the program.

PENUT-DEV, 2020-09-20:
> It seems the PIAM build step in the Makefile has been removed.
> Add the following lines in the makefile in src, just before `../rpitx' :
>
> ```
> CFLAGS_Piam   = -Wall -g -O2 -Wno-unused-variable [...]

I've done this one but the result is

Makefile:88: warning: overriding recipe for target '../piam'
Makefile:15: warning: ignoring old recipe for target '../piam'
make: No rule to make target '../am/piam.c', needed by '../piam'. Stop.

F5OEO, 2020-11-03:
piam is now replaced with csdr / sendiq script. See testpiam.sh

closed 2020-11-03 by F5OEO

Nodi-Rubrum, 2020-11-05:
Cool, we test in the next couple days.  Thanks very much for actively supporting RPiTx.