F5OEO/rpitx #168: Using sendiq random failure IQ file that is 3 seconds long returns in fraction of a second?

dachshund-digital, 2019-04-20:
I have a python class that spawns threads to do command line functions.  As part of this python class I capture the execution time, because I allow for a maximum execution time.  Typically I set the expiration time to 120 seconds, more than enough for most tasks, i.e. commands.  I also have been using RPiTx for a while, version 2 included.  On several Pi devices, it works fine. but on one of my Pi 3 model B devices, randomly sendiq returns in a fraction of a second, thus failing to transmit the IQ file as expected.  When successful the execution time is 1.5 to 2 seconds.  Test results below:

**Failure...**
[2019-04-20 01:42:31 GMT]  Fan Class Initialize...
[2019-04-20 01:42:31 GMT]  Fan Casitas Control Light (/root/IQ/Casitas/Light.iq)...
[2019-04-20 01:42:31 GMT]  Command Class Initialize...
[2019-04-20 01:42:31 GMT]  Thread Thread-45 Start...
[2019-04-20 01:42:31 GMT]  Task 'sendiq -s 250000 -f 303.875e6 -t u8 -i /root/IQ/Casitas/Light.iq'...
[2019-04-20 01:42:31 GMT]  ...Task 'sendiq -s 250000 -f 303.875e6 -t u8 -i /root/IQ/Casitas/Light.iq' (Result: 0).
[2019-04-20 01:42:31 GMT]  ...Thread Thread-44 Stop, Time 0.9804444313049316 Seconds.
[2019-04-20 01:42:31 GMT]  ...Command Class Deinitialize.
[2019-04-20 01:42:31 GMT]  ...Control Light /root/IQ/Casitas/Light.iq
[2019-04-20 01:42:31 GMT]  ...Fan Class Deinitialize.

**Success...**
[2019-04-20 01:42:38 GMT]  Fan Parameters {'Fan': 'Casitas', 'Control': 'Light'}
[2019-04-20 01:42:38 GMT]  Fan Class Initialize...
[2019-04-20 01:42:38 GMT]  Fan Casitas Control Light (/root/IQ/Casitas/Light.iq)...
[2019-04-20 01:42:38 GMT]  Command Class Initialize...
[2019-04-20 01:42:38 GMT]  Thread Thread-46 Start...
[2019-04-20 01:42:38 GMT]  Task 'sendiq -s 250000 -f 303.875e6 -t u8 -i /root/IQ/Casitas/Light.iq'...
[2019-04-20 01:42:40 GMT]  ...Task 'sendiq -s 250000 -f 303.875e6 -t u8 -i /root/IQ/Casitas/Light.iq' (Result: 0).
[2019-04-20 01:42:40 GMT]  ...Thread Thread-46 Stop, Time 1.5954720973968506 Seconds.
[2019-04-20 01:42:40 GMT]  ...Command Class Deinitialize.
[2019-04-20 01:42:40 GMT]  ...Control Light /root/IQ/Casitas/Light.iq
[2019-04-20 01:42:40 GMT]  ...Fan Class Deinitialize.

Are there any log files or such I should review to try to track this issue down?  It is really odd that the given IQ file plays in full one time and then next time (randomly) fails to play in full.  The thread code, is not aborting, it is gracefully stopping.  So it appears sendiq just fails to execute the IQ file to completion.  Oh, the IQ files are captures RF remote signals for a ceiling fan.  Running sendiq directly, I got it to fail once, so the failures are more frequent via the thread wrapper.  But on other Pi devices, I have, sendiq has been rock solid, just this one Pi where it seems to be acting up.

Guess I will reimage the SD card from a working system and see if the issue continues or not.  Weird this one is.

Jibun-no-Kage, 2022-02-18:
Just to update this... I just tested the latest RPiTx code, on Pi OS 11 (Bullseye) on a Model 2B Pi.  Using the same sendiq as shown above, and it consistently works.  I tested at both 200000 and 250000 sample rate, as well.

Jibun-no-Kage, 2022-02-18:
Tested today on 3B+, using RPiTX v2 on Pi OS 11.

Jibun-no-Kage, 2022-02-18:
Yes it does.  I posted both places in hopes of reaching a large audience.