F5OEO/rpitx #172: RPITX to SVXLINK

f5swb, 2019-05-20:
Hello everybody, 

RPITX works very well. I'm trying to use it with SVXLINK according to this : 

[http://www.pg540.org/wiki/index.php/RPITX_interface_to_SVXlink_using_GNUradio]

PTT and RX are ok but the audio output not well. 
In fact, the sound is very fast ....( x3)

the start_tx.sh is : 

echo test
nc -l 8011 | sudo /home/pi/rpitx/rpitx -i- -m IQFLOAT -f 430225 &
python /home/pi/rpitx_interface.py &
exit

Is there any change in the V2 to encode ?

Best regards.

F5OEO, 2019-05-20:
Need surely to set samplerate -s (before default was 48K, now around 260K).

f5swb, 2019-05-20:
thank you evariste, 

like this : nc -l 8011 | sudo /home/pi/rpitx/rpitx -i- -m IQFLOAT **-s 260000** -f 430225 ?

the sound is like a daffy duck and very fast ...

F5OEO, 2019-05-20:
Depend on what is the samplerate input, probably a 48Khz (check on the grc) : 
nc -l 8011 | sudo /home/pi/rpitx/rpitx -i- -m IQFLOAT -s 48000 -f 430225

f5swb, 2019-05-20:
many thanks for the quick reply, yes the sample input rate is fixed at 48Khz :)

I will try it as soon as I will be at home :) 

73's Qro.

F5SWB.

F1TZO, 2019-05-22:
Hi all,
So, to use rpitx with svxlink : 

in svxlink.conf, [Tx1]
AUDIO_DEV=udp:127.0.0.1:1233

In a script to start when svxlink swith TX : 

arecord -c1 -r48000 -fS16_LE - | nc -l -u 1233  | csdr convert_i16_f | csdr gain_ff 7000 | csdr convert_f_samplerf 20833 | sudo rpitx -i -
 -m RF -f YOUR_FREQ

I suggest to use GPIO to switch RX/TX, it's more stable than other solutions I tried

73's and many Thanks to F5OEO for this terrific soft ;-)

Michel - F1TZO

F1TZO, 2019-05-22:
Hi again all,

If you use Raspi 3 (+), you have to add in /boot/config.txt

## Specific RPI3
disable_pvt=1
force_turbo=1
arm_freq=700
gpu_freq=250
core_freq=250
sdram_freq=400

I use rpitx since 5 years with svxlink, it's work pretty fine ;-)

73's from F1TZO - Michel

F5OEO, 2019-05-22:
Thanks Michel F1TZO for this right direction...an detail info. Hope to be back on the project to set up your great svxlink work.
Best 73 Evariste

f5swb, 2019-05-23:
Many thanks to Michel F1TZO for his help and F5OEO for the fine RPITX :)
Running on a RPI 0 W very well :)

closed 2020-11-03 by F5OEO