F5OEO/rpitx #180: Rpi4?

jwcrawley, 2019-06-24:
Certainly not a complaint :) Please consider this a placeholder for today's release of the RPI4

F5OEO, 2019-06-24:
Wait for my rpi4 , will let you know.

F5OEO, 2019-06-27:
Not working out of the box. Wait for peripheral documentaion not yet released by Pi fondation.

jwcrawley, 2019-06-27:
Well nuts. I was hoping it would be a drop-in.

marius-hirjanu, 2019-08-28:
Hi F5OEO, thanks for this great tool!
Do you have any progress on this?
Don't know if it helps but they released already a preliminary document that contains the peripherals: https://www.raspberrypi.org/documentation/hardware/raspberrypi/bcm2711/rpi_DATA_2711_1p0_preliminary.pdf

developakr, 2019-12-27:
Maybe you have news on Rpi4?
we are looking forward to, thank you for a great tool!

waterwin, 2020-01-10:
Would anybody be interested to work together on creating a RPi4 working version of this software and of librpitx?

Approach:
understand the install procedure and check and when needed edit the various steps in that install process.

The thing itself looks a bit abandoned after a year of silence from the owner.

PE3ES/F4VTQ

waterwin, 2020-01-13:
The below shows the main differences between RPi3 and Ri4. What part has the most influence on the rpitx software not working?

I would expect the Broadcom chip, but why would all make/build/install steps not show any issues?

Source = https://raspberry-projects.com/pi/pi-hardware/raspberry-pi-4/rpi4-model-b-hardware-general-specifications

RPi4 Model B Hardware General Specifications
/Pi Hardware / Raspberry Pi 4 / RPi4 Model B Hardware General Specifications
Official Raspberry Pi Page
https://www.raspberrypi.org/products/raspberry-pi-4-model-b/

Key Differences To The Raspberry Pi 3
Broadcom BCM2711, Quad core Cortex-A72 (ARM v8) 64-bit SoC @ 1.5GHz

1GB, 2GB or 4GB LPDDR4-2400 SDRAM (depending on model)

2.4 GHz and 5.0 GHz IEEE 802.11ac wireless, Bluetooth 5.0, BLE

Gigabit Ethernet

2 USB 3.0 ports; 2 USB 2.0 ports.

2 × micro-HDMI ports (up to 4kp60 supported)

4-pole stereo audio and composite video port

5V DC via USB-C connector or GPIO header (minimum 2.5A power supply can be used if downstream USB peripherals consume less than 500mA in total).

Shared Specifications To The Raspberry Pi 3 B+
Same basic form factor to the RPi3, RPi 2, RPi1 Model B+.

Standard 40 pin GPIO header (fully backwards compatible with previous boards)

Power over Ethernet (PoE) enabled (requires separate PoE HAT)

Combined 3.5mm audio jack and composite video

Camera interface (CSI)

Display interface (DSI)

Micro SD card slot

waterwin, 2020-01-13:
Then we have the differences in internal memory addressing, according to here:

https://www.raspberrypi.org/forums/viewtopic.php?t=244031

Anyone know the RPi 4 base IO address (and the offset for GPIO)?

This is what I used for RPi 3:
#define BCM2708_PERI_BASE 0x3F000000
#define GPIO_BASE (BCM2708_PERI_BASE + 0x200000)

This has been mentioned over in the baremetal sub-forum here:

Raspberry Pi 4

See the post:
Raspberry Pi 4: CPU and peripherals base addresses: by gothicknight

Peripheral base is 0xFE000000.

Peripheral offsets are unchanged (as far as I can tell).

Source = https://www.raspberrypi.org/forums/viewtopic.php?f=72&t=243447#p1486302
Re: Raspberry Pi 4
Quote
Wed Jun 26, 2019 5:59 pm

CPU and peripherals base addresses can already be obtained from the Linux source files:
> https://github.com/raspberrypi/linux/bl ... pi-4-b.dts
> https://github.com/raspberrypi/linux/bl ... m2838.dtsi

I already have mine with me so so far there are a few changes:
1) Peripheral base address is now moved to 0x7E000000
2) There's a mention of ARM's GIC400 @ 0x40041000 but probably is only the distributor since the old CPU interface is still there @ 0x40000000

waterwin, 2020-01-13:
https://github.com/raspberrypi/linux/blob/rpi-4.19.y/arch/arm/boot/dts/bcm2838.dtsi

https://github.com/raspberrypi/linux/blob/rpi-4.19.y/arch/arm/boot/dts/bcm2838-rpi-4-b.dts

etc for people that really know what to look for compared to the files for the RPi3

waterwin, 2020-01-13:
and with regard to the used Broadcom chips :

https://github.com/raspberrypi/linux/blob/rpi-4.19.y/arch/arm/boot/dts/bcm2710-rpi-3-b-plus.dts

https://github.com/raspberrypi/linux/blob/rpi-4.19.y/arch/arm/boot/dts/bcm2711-rpi-4-b.dts

And in relation to the previous message the RPi-3-B relevant link
https://github.com/raspberrypi/linux/blob/rpi-4.19.y/arch/arm/boot/dts/bcm2837-rpi-3-b-plus.dts

waterwin, 2020-01-13:
Richard G Hirst added RPi4B support to his ServoBlaster and comments on **DMA** as follows:

The DMA channel used depends on the SOC type.  If the Pi has a BCM2711
(Pi4B only, at present) then it uses channel 7; otherwise it uses channel 14.

Source = https://github.com/richardghirst/PiBits/tree/master/ServoBlaster

waterwin, 2020-01-13:
See here especially as there are a lot of changes related to the RPi4 in servod.c

https://github.com/richardghirst/PiBits/commit/ff1d383a37d190ae21010dc1f3671f4d222b9158

waterwin, 2020-01-14:
The search what works and what not continues.
On RPi4
pi@raspberrypi:~ $ raspi-gpio get 4
GPIO 4: level=1 fsel=0 func=INPUT pull=UP
Pin is not changed when running rpitx ./easytest.sh and choosing one of the modulations to transmit

On RPi3, pin becomes active
pi@raspberrypi:~ $ sudo raspi-gpio get 4
GPIO 4: level=0 fsel=4 alt=0 func=GPCLK0

waterwin, 2020-01-14:
Files used of which a new RPi4 version is needed or even available
Location old file librpitx/src RPi4 updated version available here:
https://github.com/AndrewFromMelbourne/raspberry_pi_revision/blob/master/raspberry_pi_revision.h and .c

F5OEO, 2020-01-14:
Why rpitx is not yet compatible with rpi4 is not easy to explain : change in memory, DMA ,should be enough (Pifmrds), but ... Cloks are handled differenly ; as using primary PLL , still needed for tweaks to work.

waterwin, 2020-01-18:
Makefile cpu specific changes
Source : https://github.com/midwan/amiberry/blob/master/Makefile
# Specify "make PLATFORM=<platform>" to compile for a specific target.
# Check the supported list of platforms below for a ful list

# Raspberry Pi 4 CPU flags
ifneq (,$(findstring rpi4,$(PLATFORM)))
    CPUFLAGS = -mcpu=cortex-a72 -mfpu=neon-fp-armv8
endif

# Raspberry Pi 3 CPU flags
ifneq (,$(findstring rpi3,$(PLATFORM)))
    CPUFLAGS = -mcpu=cortex-a53 -mfpu=neon-fp-armv8
endif

I expect to find more in various Makefiles across github, I did not find a central point documentation of these parameters so far.

goutx2, 2020-04-19:
BCM2711 peripherals

https://github.com/raspberrypi/documentation/tree/master/hardware/raspberrypi/bcm2711
 We are waiting   rpitx for rpi4

goutx2, 2020-04-26:
Any progress  for porting to rpi4 ?

bootrino, 2020-04-28:
Hi Evariste - fabulous software thankyou.

I'm curious - do you think that it's not possible for rpitx to work with Raspberry Pi 4?  Or are there other people who might be able to help to resolve the issues - maybe there's people at Raspberry Pi Foundation who might be willing to help get the issues resolved - would you mind if I asked them if they can help?

VK3ZTE, 2020-05-06:
I also am deeply appreciative of F5OEO and his amazing efforts with rpitx.  I have it running on a couple of RPi3s but have issues with running out of memory after several hours operation which necessitates rebooting.  Would therefore like to try running it on RPi4 with 4MB memory.  Unfortunately, I don't have the IT skills to usefully contribute to porting it across from one platform to the other.  Does anyone have news re developments?

makeworld-the-better-one, 2020-05-13:
[This README](https://github.com/markondej/fm_transmitter#raspberry-pi-4) mentions how for their RPi FM code, the single biggest change is using GPIO pin 21 instead of 4. I suspect pin 4 is too close to some hardware emitting EM frequencies. It also suggests disabling CPU freq. scaling, and keeping transmit frequencies to below 93 MHz. Hope this helps, best of luck.

goutx2, 2020-05-21:
Hello Evariste juste tell as wich part we have to change to make rpitx compatible  with rpi4 ,we have waiting for long time but there no news

F5OEO, 2020-07-23:
To beta testers : A fresh version of librpitx (branch pi4) is available for test. Seems ok up to 100Mhz. Need more tweaks.

makeworld-the-better-one, 2020-07-23:
@F5OEO I'm not seeing the branch called `pi4` in this repo. Happy to hear you've gotten stuff working though!

VK3ZTE, 2020-07-23:
Marvellous news.  

 

Many thanks

 

Grant

VK3ZTE

romikette, 2020-08-19:
Could I get on that beta testing list? I'd love to try out the beta branch!
Thanks!

veliztick, 2020-10-02:
Could I get on that beta testing list?

romikette, 2020-10-02:
It's the librpitx that is in beta for the rpi4. I thought that it was rpitx itself aswell. F5OEO did leave clues to this ..
I found this [link of him explaining how to do it](https://groups.io/g/rpitx/topic/rpitx_on_pi4_beta/75747197) just yesterday.

> Hi All,
After a time of silence, I am back and announce that a beta version is available for testing on PI4.
You need to checkout the librpitx pi4 branch , git pull latest master rpitx recompile.
Please add line 
force_turbo=1
on your /boot/config.txt
Feedbacks welcomed !
Evariste

_this next part I presume you do it after initial install so it can update_
> In rpitx/src you should have a librpitx folder.
Go in this directory and make a "git pull" and "git checkout pi4" then "make -C src"
Now under rpitx/src : "git pull " and "make"

F5OEO, 2020-11-03:
Pi4 release today !

closed 2020-11-03 by F5OEO