F5OEO/rpitx #182: POCSAG message incorrect in function 0 numeric mode

mdozier9, 2019-07-30:
I am trying to send a number (function 0) message, but I seem to keep getting an incorrect message.  When I try the message using function 1 or 2, it appears to work fine.  I've tried inverting as well with no luck.  Here is my command:
printf "123456A:12345" | sudo ./pocsag -f 434000000.  

When monitoring from multimon, I get this:
POCSAG1200: Address:  123456  Function: 0  Numeric: 139- 86530.  

The address is always correct, and the message is always consistent, but certainly not the "123456".  I've tried setting the function code using the command line switch as well, and the output is the same.  Here is the debug output of my command as well:
Fifo Size = 12000, Size = 35, Baud rate = 1200
Channel 14 CBSize 24006 UsermemSize 12003
Using mbox device /dev/vcio.
816204 Size NUM PAGES 200 PAGE_SIZE 4096
BufferSize 12003 , cb 24006 user 12003
Info:NTP find offset -81333 freq -513616 pps=0 ppm=-7.837158
Info:NTP find offset -81333 freq -513616 pps=0 ppm=-7.837158
Info:NTP find offset -81333 freq -513616 pps=0 ppm=-7.837158
Info:NTP find offset -81333 freq -513616 pps=0 ppm=-7.837158
Freq PLL no 5= 1301986454
Found PLL solution for frequency 434.0MHz : divider:3 VCO: 1302.0MHz
Master PLLC Locked
Info:NTP find offset -81333 freq -513616 pps=0 ppm=-7.837158
Freq PLL no 6= 500003924
PCM prediv = 102
PCM clk=4084 / 4093
FSK Origin fsel 0
0101010101010101 0101010101010101
0101010101010101 0101010101010101
0101010101010101 0101010101010101
0101010101010101 0101010101010101
0101010101010101 0101010101010101
0101010101010101 0101010101010101
0101010101010101 0101010101010101
0101010101010101 0101010101010101
0101010101010101 0101010101010101
0101010101010101 0101010101010101
0101010101010101 0101010101010101
0101010101010101 0101010101010101
0101010101010101 0101010101010101
0101010101010101 0101010101010101
0101010101010101 0101010101010101
0101010101010101 0101010101010101
0101010101010101 0101010101010101
0101010101010101 0101010101010101
1000001100101101 1110101000100111
1111100001110110 1111101001110100
0011100110110010 0110010111100100
0111010010101001 1111110100111110
1000010101110110 0011111001101000
1000010101110110 0011111001101000
1000010101110110 0011111001101000
1000010101110110 0011111001101000
1000010101110110 0011111001101000
1000010101110110 0011111001101000
1000010101110110 0011111001101000
1000010101110110 0011111001101000
1000010101110110 0011111001101000
1000010101110110 0011111001101000
1000010101110110 0011111001101000
1000010101110110 0011111001101000
1000010101110110 0011111001101000
Symbols=1120
FSK burst end Tx

It is something I'm doing wrong, or is there an issue with function zero?
Tnx, Mike, KB4NKO

mdozier9, 2019-08-02:
Upon further investigation, it appears that the code is encoding a funtion 0 (numeric message) as 7 bits as opposed to 4 bit encoding as called for in the protocol spec.  For example when we send a numeric message of "123", its getting encoded as 1000110	0100110	1100110 (LSB is to the left).  Note the 7-bit ASCII value encoding rather than 4-bit BCD.  When a numeric pager tries to decode that, it will look for 4-bit encoding and it will read it as such:  1000	1100	1001	1011	0011	0100.  A numeric only pager client (demand load switch in my case) will interpret this as "139- ", and not "123".  Obviously, a device expecting the message "123" would not work.  Most of the demand load switches out on ebay are function zero, numeric only devices.  I am trying to get this library to control electrical loads behind these switches.  I'm not very savvy in coding, so any help on how to fix this would be greatly appreciated.  Tnx, Mike, KB4NKO

mdozier9, 2019-08-03:
Changing line 85 to: #define TEXT_BITS_PER_CHAR 4 works. However it does break text messages.  Perhaps some logic can be included to look at the SetFunctionBits and then set TEXT_BITS_PER_CHAR accordingly (4 bits for function 0, else 7 bits for all others).

cuddlycheetah, 2019-09-11:
Try it with my recent commit https://github.com/F5OEO/rpitx/commit/528c18be903314beec25da2af7448e9e209d1640

and run it with the argument "-n", so it will write 5 digits per word.

mdozier9, 2019-09-12:
Works like a charm using -n.   So is it not a safe assumption that a function code 0 and "numeric encoding" are that same?  For some reason I always thought they were.  I guess I need to go back and read the protocol spec. lol   Thanks a bunch!  Mike, KB4NKO

cuddlycheetah, 2019-09-12:
Function Code 0 ,1,2,3 are just „Function Codes“, they dont describe the Data.
Theorethically it would be possible to transmit numeric Messages over the ascii function, but you have to know that there will be a problem, because ascii Encoding takes 7 bits per char, filling 20 bits per word. And the numeric Messages are encoded via 4 bits BCD. 1 Word can take up to 5 Digits.

Von: mdozier9
Gesendet: Donnerstag, 12. September 2019 02:40
An: F5OEO/rpitx
Cc: Cuddly Cheetah; Comment
Betreff: Re: [F5OEO/rpitx] POCSAG message incorrect in function 0 numeric mode(#182)

Works like a charm using -n. So is it not a safe assumption that a function code 0 and "numeric encoding" are that same? For some reason I always thought they were. I guess I need to go back and read the protocol spec. lol Thanks a bunch! Mike, KB4NKO