F5OEO/rpitx #183: Pocsag problem with Nec21A (Skyper)

EU4DGC, 2019-08-15:
Hey. I am sending such a command.
echo -e "71632:Hello world. Testmessage!" | sudo ./pocsag -f 159.050e6
At the same time, the pager changes the icon that sees the signal, but no messages arrive.
Any idea?

EU4DGC, 2019-08-20:
Thanks for the answer and advice.  I tried different combinations, but unfortunately nothing comes of it.  I hear a signal in the radio station, the reception icon on the pager changes.  But messages do not come.

pyrog, 2019-08-21:
Did you tried different speeds and configurations ?
Do you know the settings of your pager ?

Mine display frequency and baud rate at startup.

If you have the good frequency and baud rate, you "just" have to send 2
messages:
one with normal "polarity" and another "inverted".

--
Yves

EU4DGC, 2019-08-21:
I tried different modes and commands.  The frequency is exactly right, because  the network icon is changing.  I tried to change the speed, inversion too.

VasylSamoilov, 2019-10-26:
Same problem with Motorola advisor linguist. 
Tried nearly all possible combinations, nothing seems to work on a pager.
Shows several "antenna" icons, but no reaction at all.
Tried raspbian stretch and buster.
Tuned with RTL-SDR, frequency seems to be right.
FM radio transmission on the same frequency works quite good.
tried reflashing pager to 512 and 1200 speed rate.
`printf "0704472:YOURCALL1" | sudo ./pocsag -f "161125000" -r 512 -b 3
printf "0704472:YOURCALL2" | sudo ./pocsag -f "161125000" -r 1200 -b 3
printf "0704472:YOURCALL3" | sudo ./pocsag -f "161125000" -r 2400 -b 3

printf "0704472:YOURCALL1" | sudo ./pocsag -f "161125000" -r 512 -b 3 -i
printf "0704472:YOURCALL2" | sudo ./pocsag -f "161125000" -r 1200 -b 3 -i
printf "0704472:YOURCALL3" | sudo ./pocsag -f "161125000" -r 2400 -b 3 -i

printf "0704472:YOURCALL\n1999841: Hello world" | sudo ./pocsag -f "161125000" -i
echo "0704472:0000001234567789" | sudo ./pocsag -f "161125000" -n -i
printf "0704472:YOURCALL" | sudo ./pocsag -f "161125000"
printf "0704472:YOURCALL3" | sudo ./pocsag -f "161125000"
printf "1020304:YOURCALL2\n1999641: Hello my world" | sudo ./pocsag -f "161125000"
echo "0704472:0000234567789" | sudo ./pocsag -f "161125000" -n
printf "704472D:1234567" | sudo ./pocsag -f "161125000" -d
printf "1999841A:10101010" | sudo ./pocsag -f "161125000" -d

`

mdozier9, 2019-10-26:
Is it possible that your pager has been programmed with a different capcode than what is displayed on the label?  Many models are programmable, so possibly it has been reprogrammed.

VasylSamoilov, 2019-10-26:
No, I got it connected to a programmer. Numbers are correct. I also tried capturing with sdr# and virtual audio cable to decoder and it only able to decode 2400 message and detect 1200 with zero successful decodes. 512 it can't even see. Maybe it's something to do with frequency, and not protocol?

VasylSamoilov, 2019-10-26:
I switched pager to 2400 but no luck, regretfully.

VasylSamoilov, 2019-10-27:
It was a faulty pager, got another one working on 512 baud.

anjodoazar, 2019-11-12:
Examples:
NEC pa3py-20b
printf "0116392:Hello world. Testmessage!" | sudo ./pocsag -f "164925000" -r 512
"The date stopped on 12/2013.
You can't get it right from that date."
Motorola scriptor lx2
printf "2045784:Hello world. Testmessage" | sudo ./pocsag -f "164925000" -r 512
bipper
From Portugal.

Marv0123, 2020-12-31:
i have the same problem with an motorola scriptor lx4 and an motorla lifestyle plus...    i think it's not possible that all pagers are faulty?!   The PDW decoding Software recognises the Signal properly...

Marv0123, 2021-01-07:
does anyone know how to fix this problem ?

VasylSamoilov, 2021-01-08:
The problem is, probably, with the pager radio. It's tuned to another frequency, maybe inverted or just plain faulty. Did you programmed pager yourself to proper CAP codes and speed? try writing a script that will send pages to known good cap code and speed across multiple frequencies.

Marv0123, 2021-01-09:
Well, the baud rate is definitely 512 baud, i have also tested 1200 and 2400. In the case of 1200 and 2400 the "no signal" icon is still there. At 512 baud it is gone so the pager recognises the Signal....    So the problem could only stuck at the RIC-CAPCODE? Right?  The frequency is also right at 162.050Mhz....

VasylSamoilov, 2021-01-11:
You can't be sure about frequency unless you use oscilloscope, and can't be sure about CAP code and speed unless you programmed it yourself. Also try "-i" option.

Marv0123, 2021-01-11:
Now i think i know the issue, i put un the CAP Code 9114419 and it decodes to 0725811.... the program makes the mistake ..

EU4DGC, 2021-06-07:
Hello everybody.  It turned out to send a message to the Nec 21A pager.  As they wrote, the whole problem was in the wrong CAP code.  Below is a link on how to program a pager using Arduino.
https://qsy.by/programmator-dlya-pejdzhera-nec-21a-na-arduino/