F5OEO/rpitx #184: Raspian Buster?

clemibunge, 2019-08-22:
Rpitx not working on Raspi zero W with Raspian buster

dade80vr, 2019-09-02:
Seems not works also on Raspi Pi 3B+ with Buster!
Any thoughs?

Murphyv310, 2019-10-10:
No transmit on pi4 program seems ok but no results. Any ideas

Murphyv310, 2019-10-19:
Hi.
Just wondering if there is a fix for this? I purchased a Pi4 specifically for rpitx to feed a display of early FM radio's and modern ones with RDS. Disappointed that my money is wasted.

Murphyv310, 2019-12-22:
Hi.
Is there still nothing on this issue?

Anjum9694, 2019-12-26:
RPiTX running raspbian buster headless on 3B+ works for me

Murphyv310, 2019-12-26:
Tried this on two Pi4B with buster which results in no RF output.

clemibunge, 2019-12-26:
@Murphyv310 maybe you have more lock using https://github.com/ChristopheJacquet/PiFmRds with your Pi4 it says it supports pi 4

Murphyv310, 2019-12-26:
Seems that PiFmRd has been configured for the new Chip set in the Pi4 whereas rpitx hasn't, can't see how anyone can get this to work on a Pi4..........

makeworld-the-better-one, 2020-05-13:
Is this still broken?

dachshund-digital, 2020-07-12:
I have almost every model of Pi, testing, so far none have worked with buster for me.

Bleach665, 2020-07-12:
For me work fine on clean install of 2020-05-27-raspios-buster-armhf.img.

Murphyv310, 2020-07-12:
Will try this later. So we are suggesting that the 27th May release of Raspberrry Pi OS 32Bit and a Pi version 4B will now let RPiTX work?
Like I say I'll try this later on.

Bleach665, 2020-07-12:
Sorry, I didn't specify what I used Raspberry PI Zero.