F5OEO/rpitx #187: Warning samplerate too high decimation by 2 will be performed

matrix199621, 2019-10-07:
hi when i record an IQ file using rtl sdr and send it back it says  Warning samplerate too high decimation by 2 will be performed and i dont get anything on the device am trying to make the replay attack on i truly wish you'll help me ! the frequency is 433mhz and the second one is 315 mhz knowing that i have tried everything i found on the issues guide and am using a raspberry pi 3 model B+

urosch, 2019-11-18:
same thing here, using raspberry 3b+ (Buster)and I get the same message, the transmitted signal is longer (in ms) than original one. Any idea why this happens?

matrix199621, 2019-11-21:
the developer didnt answer i even contacted him on twiter.....i needed it for a final year project

seadolphine0, 2020-02-08:
Here is a solution that worked for me:
1. Delete the installed version.
2. Clone the repo again.
3. Before running install.sh, open src/sendiq.cpp file.
4. Search for #define MAX_SAMPLERATE 200000
5. Change it to 250000
6. Run install.sh as usual.

It worked for me on Kali-2019.1-rpi3-nexmon

PENUT-DEV, 2020-09-20:
> Here is a solution that worked for me:
> 
> 1. Delete the installed version.
> 2. Clone the repo again.
> 3. Before running install.sh, open src/sendiq.cpp file.
> 4. Search for #define MAX_SAMPLERATE 200000
> 5. Change it to 250000
> 6. Run install.sh as usual.
> 
> It worked for me on Kali-2019.1-rpi3-nexmon

is this still working on kali 2020? i kept getting the same error after I change the Sample rate in sendiq.cpp file

wstlxx, 2023-07-17:
you might want to check the path to the IQ file because it looks like this "samplerate too high decimation by 2 will be performed" when the file is nowhere to be found. Usually saved in another directory.