F5OEO/rpitx #198: Pocsag will not work

Marv0123, 2020-02-04:
Hello, 
i am trying to write a message to a motorola pager frequency is 162.050 Mhz RIC 0550602, 

I use the command: echo -e "0550602: Testmessage" | sudo ./pocsag -f 162.050e6 -r 512

The pager seems to recognize the signal but will not beep or show anything..

What's wrong?

Yours, Max

pyrog, 2020-02-05:
Hi,

> I use the command: echo -e "0550602: Testmessage" | sudo ./pocsag -f 162.050e6 -r 512
> 
> The pager seems to recognize the signal but will not beep or show anything..
> 
> 
Try the -i option.

BR,

—
Yves

Marv0123, 2020-02-10:
no also the i-option will not work... hmm.

F5OEO, 2020-11-03:
Pocsag protocol and not directly related to rpitx

closed 2020-11-03 by F5OEO

Marv0123, 2020-11-04:
Which raspberry model do u use ?