F5OEO/rpitx #203: Increase Chirp Carrier moves around 434 ! is that possible ???

khanfar, 2020-04-05:
is that possible to make chirp carrier move like from 410 mhz to 420 mhz ???? is there way to increase chirp carrier move bandwidth ???

khanfar, 2020-04-05:
like move in cycle forward and backward loop between 410-420 mhz . is that possible ? or add in next version update ?

khanfar, 2020-04-06:
​#​include​ ​<​unistd.h​>​
#​include​ ​"​../librpitx/src/librpitx.h​"​
#​include​ ​<​unistd.h​>​
#​include​ ​"​stdio.h​"​
#​include​ ​<​cstring​>​
#​include​ ​<​signal.h​>​
#​include​ ​<​stdlib.h​>​

​bool​ running=​true​;

​void​ ​SimpleTestDMA​(​uint64_t​ Freq)
{
        

        ​int​ SR=​1000​;
        ​int​ FifoSize=​4096​;
        ngfmdmasync ​ngfmtest​(Freq,SR,​14​,FifoSize);
        ​for​(​int​ i=​0​;running;)
        {
                ​//​usleep(10);​
                ​usleep​(FifoSize*​1000000.0​*​3.0​/(​4.0​*SR));
                ​int​ Available=ngfmtest.​GetBufferAvailable​();
                ​if​(Available>FifoSize/​2​)
                {        
                        ​int​ Index=ngfmtest.​GetUserMemIndex​();
                        ​//​printf("GetIndex=%d\n",Index);​
                        ​for​(​int​ j=​0​;j<Available;j++)
                        {
                                ​//​ngfmtest.SetFrequencySample(Index,((i%10000)>5000)?1000:0);​
                                ngfmtest.​SetFrequencySample​(Index+j,(i%SR)/​10.0​);
                                i++;
                        
                        }
                }
                
                
        }
        ​fprintf​(stderr,​"​End​\n​"​);
        
        ngfmtest.​stop​();
        
}

​static​ ​void​
​terminate​(​int​ num)
{
    running=​false​;
        ​fprintf​(stderr,​"​Caught signal - Terminating​\n​"​);
   
}

​int​ ​main​(​int​ argc, ​char​* argv[])
{
        
        ​if​(argc<​3​) {​printf​(​"​Usage : pichirp Frequency(Hz) Bandwidth(Hz) Time(Seconds)​\n​"​);​exit​(​0​);}
        
        ​float​ Frequency=​atof​(argv[​1​]);
        ​float​ Bandwidth=​atof​(argv[​2​]);
        ​float​ Time=​atof​(argv[​3​]);

         ​for​ (​int​ i = ​0​; i < ​64​; i++) {
        ​struct​ ​sigaction​ sa;

        ​std::memset​(&sa, ​0​, ​sizeof​(sa));
        sa.​sa_handler​ = terminate;
        ​sigaction​(i, &sa, ​NULL​);
    }
        
        ​int​ SR=​200000​;
         
        ​int​ FifoSize=​4096​;
        ngfmdmasync ​ngfmtest​(Frequency,SR,​14​,FifoSize);
        ​float​ Step=Bandwidth/Time;        ​//​Deviation Hz by second ​
        ​float​ StepWithSR=Step/(​float​)SR;
        ​int​ NbStepWithSR=Time*SR;
        ​float​ FrequencyDeviation=​0​;
        ​int​ count=​0​;
        ​for​(​int​ i=​0​;running;)
        {
                
                ​usleep​(FifoSize*​1000000.0​*​3.0​/(​4.0​*SR));
                ​int​ Available=ngfmtest.​GetBufferAvailable​();
                ​if​(Available>FifoSize/​2​)
                {        
                        ​int​ Index=ngfmtest.​GetUserMemIndex​();
                        
                        ​for​(​int​ j=​0​;j<Available;j++)
                        {
                                
                                ngfmtest.​SetFrequencySample​(Index+j,Bandwidth*​0.5​*​sin​(​2​*​3.1415​*(​float​)count/(​float​)NbStepWithSR));
                                count++;
                                ​if​(count>NbStepWithSR) count=​0​;
                        
                        }
                        
                }
                ​fprintf​(stderr,​"​Freq=%f​\n​"​,Bandwidth*​0.5​*​sin​(​2​*​3.1415​*(​float​)count/(​float​)NbStepWithSR));
                
        }
        ​fprintf​(stderr,​"​End​\n​"​);
        
        ngfmtest.​stop​();

        
        
        
}

khanfar, 2020-04-06:
What shoud I edit in chirp to increase moving bandwidth ?

khanfar, 2020-04-06:
When I chose frequency in chirp like 434 mhz and start it , Iim looking the signal move +5mhz forward then go backward -5mhz from center (434) and so on ... on loop ... backward and forward...

F5OEO, 2020-11-03:
sudo ./pichirp 435e6 10e6 10

closed 2020-11-03 by F5OEO