F5OEO/rpitx #206: On Raspberry PI Zero W (Buster) stop transmitting after less than a second

algorni, 2020-05-13:
Just running on raspi pi zero w, connected in ssh over wifi.

pi@raspi-zero-w:~/radio/rpitx $ cat /etc/os-release
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

pmbdk, 2020-06-09:
I am running exactly the same setup using the latest raspberry pi os lite and it works fine here

algorni, 2020-06-09:
BTW i've similar issue also on PI 3     not exactly the same... but similar....   some time transmit some other no,

dachshund-digital, 2020-07-12:
Same here, on Pi3 with buster, it seems to take at least 2 tries to get something to transmit.

Bleach665, 2020-07-13:
Same on RPi Zero, clean install of `2020-05-27-raspios-buster-armhf.img`.
After random time the transmitting breaks.

dachshund-digital, 2020-07-15:
I did the same as you, testing on Pi ZeroW, it crashed or hung my Pi.  I had to pull the power to reset.  It was a clean build with latest Raspberry Pi OS image as you noted above.  I did get it to work randomly on a Pi 3 model B, but if will fail in less than 1 seconds maybe 10 times before it works once.  Same clean latest Raspberry Pi OS build as noted above.

dachshund-digital, 2020-07-15:
More testing, this time on Stretch rather than Buster... the quick fail happens on Stretch as well, just not as often as on Buster.  Using a Pi Zero W, it does not lock up or crash like it does using Buster.

Ghaz75, 2020-07-17:
On my RPi3 same problem. Only SSB mod transmits for around 60 seconds rest all mods stops after 3 to 4 seconds. Is there any workaround to address this issue?

dachshund-digital, 2020-07-19:
Unfortunately, I have only have one Pi3 that seems to work, and only with Stretch.  At one time I had 4 different Pi devices working with RPiTx, various models.  But They were all running Jessie.  As I have moved to Stretch and then Buster, various issues have appeared such that my Pizeros and even Pi2s don't work with RPiTx on any recent Pi OS.

Un-ka, 2021-07-30:
I had the same problem. Try running in the background.
`&`

user421921, 2021-08-05:
I have the same thing, it runs but it transmits for less than a second, does anyone has a fix?

Ghaz75, 2021-08-06:
What I have noticed is it is having problem on all raspberry pi modules  running raspbian buster. Only one OS in which it worked flawlessly is Raspbian Jessie.

Un-ka, 2021-08-06:
I use Raspberry pi zero w, Rasbian buster aka Debian 9.
Console - JuiceSSH.
The following command works for me:
```bash
pifmrds -freq 103.0 -audio /home/pi/Music/v* -ps FBK -rt FBK &
```
Where `pifmrds` is _alias_ or _path_ to executive file.

dachshund-digital, 2021-08-14:
I am trying to use sendiq, which fails.

8cH9azbsFifZ, 2023-07-06:
Same problem here with Zero W and latest raspian.

elektronika-ba, 2024-03-19:
Same problem here with freshly installed Raspberry Pi OS Lite (32 bit), Debian Bookworm 2024-03-15.
Hardware: Raspberry PI 3 Model B v1.2