F5OEO/rpitx #209: not able to record frequency in iq file please help..

yashgupta1262, 2020-07-03:
command i am using ..
rtl_sdr -s 250000 -g 42 -f 433937500 car_remote.iq

error
e.iq
Found 1 device(s):
  0:  Realtek, RTL2838UHIDIR, SN: 00000001

Using device 0: Generic RTL2832U OEM
Found Rafael Micro R820T tuner
Exact sample rate is: 250000.000414 Hz
[R82XX] PLL not locked!
Sampling at 250000 S/s.
Tuned to 433937500 Hz.
Tuner gain set to 42.10 dB.
Failed to open car_remote.iq

its says failed to open car_remote.iq 
please help me to solve thanks in advance...

andimik, 2020-07-03:
Have you tried `-` (for stdout) instead of the file name?

http://manpages.ubuntu.com/manpages/trusty/man1/rtl_sdr.1.html

yashgupta1262, 2020-07-03:
rtl_sdr -s 250000 -g 42 -f 433937500 -  @andimik  i make this command its worked but it not save any output file now i am confused how to replay it ... i am using this   sudo ./sendiq -s 250000 -f 433937500 - but it not working please help...
thanku so much for helping sir....

andimik, 2020-07-03:
Do you have write access in the current folder?

F5OEO, 2020-11-03:
Not related to rpitx itself.

closed 2020-11-03 by F5OEO