F5OEO/rpitx #212: RPiTx now fails on Stretch and Buster Pi OS distributions, randomly crashing Pi Model 3B and Model 3B+

dachshund-digital, 2020-07-17:
After extensive RPiTx now fails on Stretch and Buster Pi OS distributions, randomly crashing Pi Model 3B and Model 3B+ with no errors or faults, just hangs the Pi devices, locks them up, or signal is never generated.  I have only consistently been able to get RPiTx to work on one of my Pi devices, an older Pi Model 3B that has a very old OS instance, that has been updated over and over, but not having the OS completely replaced our right.  So the one working Pi, OS is likely and incomplete distribution upgrade some time ago.

Really did love using RPiTx while it was stable and actively supported.  Oh, and it fails badly on all of my Pi 4 devices, I have tried all the suggestions here, but no joy so far.  Hope one or some of the the original maintainers pick up supporting this again.  There were a lot of fans of this solution.

dachshund-digital, 2020-07-21:
I went all the way back to Jessie (Raspbian 8), and install RPiTx, and I still see the aborted sendiq attempts, that we now see on Stretch and Buster.  So the issue is core to the kernel or something that is very low level in the OS, if not an updated library.  The only things I did to the Jessie image, was raspi-config, rpi-update, apt update, and apt upgrade.  Looks like RPiTx is dead and long forgotten, since there does not seem to be any active maintainers?

F5OEO, 2020-11-03:
As soon as I can close this issue, rpitx is not dead : moreover it is tested on latest Raspbian OS.

closed 2020-11-03 by F5OEO

Nodi-Rubrum, 2020-11-05:
Cool, we test in the next couple days. Thanks very much for actively supporting RPiTx.