F5OEO/rpitx #214: No audio/sound on Raspberry Pi 2 after using sendiq

JohnDN90, 2020-07-22:
**_Hardware_**
Device: Raspberry Pi 2 Model B
rpitx version: master commit dd69751
Operating System: DietPi v6.31.2
Audio Output: 3.5mm Jack
`Linux hostname 5.4.51-v7+ #1326 SMP Fri Jul 17 10:45:25 BST 2020 armv7l GNU/Linux1`

_**Issue**_
I boot the Raspberry Pi 2 and confirm that I have audio.  Next, I run the following command.

`sudo sendiq -s 250000 -f 315.00e6 -t u8 -p 7.00 -i record.iq`

I'll hear a quick "popping" sound from the speakers (similar to when you plug or unplug something from an audio jack).  I try playing any type of audio/sound again from the Raspberry Pi and it doesn't work.  Once I reboot the Raspberry Pi audio is working again.

I've seen a few other audio problems mentioned in various comments in a couple other issues, but haven't seen a solution.  Has anyone else experience this issue and found a solution?  Alternatively, does anyone have a workaround, such as another command to "restart" the audio after running the `sendiq` command without having to reboot?

F5OEO, 2020-07-23:
rpitx use pwm clock to send constant samplerate. Pwm on raspberry is also used for audio output, so it is normal that it breaks your audio. Workaround is to use a usb audio dongle for audio output.

JohnDN90, 2020-07-23:
Thank you for clearing that up.

I assume pwm control is taken over durnig the  "initialization" of rpitx when one of its commands are called.  Is there anyway to return pwm control back to audio after rpitx exits, for example manually running a set of commands after rpitx has exited?

F5OEO, 2020-11-03:
Not easy..prefere a reboot as rpitx move a lot of registers , difficult to get back to standard pwm audio

closed 2020-11-03 by F5OEO