F5OEO/rpitx #215: Can not compile on RPi4 running Ubuntu 64bit.

FalcoGer, 2020-08-03:
The readme say it works in "beta mode". But there are no instructions on how to compile in beta mode. I tried to compile as is, but I had to change the csdr/Makefile

```Makefile
# PARAMS_NEON = -mfloat-abi=hard -march=armv7-a -mtune=cortex-a8 -mfpu=neon -mvectorize-with-neon-quad -funsafe-math-optimizations -Wformat=0 -DNEON_O>
PARAMS_NEON = -mcpu=cortex-a53 -march=armv8-a+crc+crypto -fomit-frame-pointer -funroll-loops -Wformat=0
#tnx Jan Szumiec for the Raspberry Pi support
# PARAMS_RASPI = -mfloat-abi=hard -mcpu=arm1176jzf-s -mfpu=vfp -funsafe-math-optimizations -Wformat=0
PARAMS_RASPI = -mcpu=native -mfpu=vfp -funsafe-math-optimizations -Wformat=0
```

Compiler complained about unrecognized flags `-mfloat-abi=hard -mfpu=neon -mvectorize-with-neon-quad` so I tossed them out.

I tried to compile, but it failed with a linker problem with -lbcm_host not being found.
I had to fetch raspberrypi/userland and make bcm_host and install that, otherwise csdr wouldn't compile with a linker error.

Then I got a bunch of errors in dvbsenco8.s. I replaced `@` with `#` but it still is very much not happy.

After all this I get these errors:

```text
~/repos/rpitx/src>make
cc -Wall -g -O2 -Wno-unused-variable -c -o dvb/dvbsenco8.o dvb/dvbsenco8.s
dvb/dvbsenco8.s: Assembler messages:
dvb/dvbsenco8.s:67: Error: unknown mnemonic `push' -- `push {r14}'
dvb/dvbsenco8.s:71: Error: unknown mnemonic `pop' -- `pop {r14}'
dvb/dvbsenco8.s:72: Error: unknown mnemonic `bx' -- `bx lr'
dvb/dvbsenco8.s:99: Error: unknown mnemonic `push' -- `push {r1,r2,r14}'
dvb/dvbsenco8.s:101: Error: operand 1 must be an integer register -- `ldr r1,=workbuff#:output to workbuff'
dvb/dvbsenco8.s:104: Error: operand 1 must be an integer register -- `ldr r0,=workbuff'
dvb/dvbsenco8.s:107: Error: operand 1 must be an integer register -- `ldr r0,=workbuff'
dvb/dvbsenco8.s:108: Error: unexpected characters following instruction at operand 1 -- `bl interleave#:returns with r0=pointer to an output packet'
dvb/dvbsenco8.s:113: Error: unknown mnemonic `pop' -- `pop {r1,r2,r14}'
dvb/dvbsenco8.s:114: Error: unknown mnemonic `bx' -- `bx lr'
dvb/dvbsenco8.s:138: Error: unknown mnemonic `push' -- `push {r0,r1}'
dvb/dvbsenco8.s:139: Error: operand 1 must be an integer register -- `ldr r1,=energytableindex'
dvb/dvbsenco8.s:140: Error: operand 1 must be an integer register -- `mov r0,#0'
dvb/dvbsenco8.s:141: Error: operand 1 must be an integer register -- `str r0,[r1]'
dvb/dvbsenco8.s:143: Error: operand 1 must be an integer register -- `ldr r1,=packetcount'
dvb/dvbsenco8.s:144: Error: operand 1 must be an integer register -- `mov r0,#0'
dvb/dvbsenco8.s:145: Error: operand 1 must be an integer register -- `str r0,[r1]'
dvb/dvbsenco8.s:147: Error: unknown mnemonic `pop' -- `pop {r0,r1}'
dvb/dvbsenco8.s:148: Error: unknown mnemonic `bx' -- `bx lr'
dvb/dvbsenco8.s:155: Error: unknown mnemonic `push' -- `push {r0,r1,r2,r3,r4,r5,r6,r7,r8}'
dvb/dvbsenco8.s:156: Error: operand 1 must be an integer register -- `ldr r4,=packetcount'
dvb/dvbsenco8.s:157: Error: operand 1 must be an integer register -- `ldr r4,[r4]#:packetcount'
dvb/dvbsenco8.s:158: Error: operand 1 must be an integer or stack pointer register -- `and r4,#(NIPACKETS-1)#:modulo'
dvb/dvbsenco8.s:159: Error: operand 1 must be an integer register -- `ldr r2,=energytableindex'
dvb/dvbsenco8.s:160: Error: operand 1 must be an integer register -- `ldr r5,[r2]#:energytableindex'
dvb/dvbsenco8.s:162: Error: operand 1 must be an integer register -- `mov r6,#0x47'
dvb/dvbsenco8.s:163: Error: operand 1 must be an integer register -- `ands r4,#7'
dvb/dvbsenco8.s:164: Error: unknown mnemonic `moveq' -- `moveq r6,#0xb8'
dvb/dvbsenco8.s:165: Error: unknown mnemonic `moveq' -- `moveq r5,#0#:reset energytableindex'
dvb/dvbsenco8.s:166: Error: operand 1 must be an integer register -- `strb r6,[r0]#:save sync byte in the packet'
dvb/dvbsenco8.s:169: Error: operand 1 must be an integer register -- `mov r6,#188/4'
dvb/dvbsenco8.s:170: Error: operand 1 must be an integer register -- `ldr r7,=energytable'
dvb/dvbsenco8.s:172: Error: operand 1 must be an integer register -- `ldr r8,[r0],#4#:get a word to process'
dvb/dvbsenco8.s:173: Error: operand 1 must be an integer register -- `ldr r3,[r7,r5]#:get a word from the energy table'
dvb/dvbsenco8.s:174: Error: operand 1 must be an integer or stack pointer register -- `eor r8,r3#:r8=r8^r3'
dvb/dvbsenco8.s:175: Error: operand 1 must be an integer or stack pointer register -- `add r5,#4#:increment energytableindex'
dvb/dvbsenco8.s:176: Error: operand 1 must be an integer register -- `str r8,[r1],#4#:store it back and move to next'
dvb/dvbsenco8.s:177: Error: operand 1 must be an integer register -- `adds r6,#-1#:decrement loop counter'
dvb/dvbsenco8.s:180: Error: operand 1 must be an integer register -- `str r5,[r2]#:save energytableindex'
dvb/dvbsenco8.s:183: Error: unknown mnemonic `pop' -- `pop {r0,r1,r2,r3,r4,r5,r6,r7,r8}'
dvb/dvbsenco8.s:184: Error: unknown mnemonic `bx' -- `bx lr'
dvb/dvbsenco8.s:213: Error: unknown mnemonic `bx' -- `bx lr'
dvb/dvbsenco8.s:217: Error: unknown mnemonic `push' -- `push {r0,r1,r2,r3,r4,r5,r6,r7,r8,r9,r10,r11,r12}'
dvb/dvbsenco8.s:220: Error: operand 1 must be an integer or stack pointer register -- `add r1,r0,#188#:r1=pointer to 16 RS bytes at the end of 188 bytes'
dvb/dvbsenco8.s:222: Error: operand 1 must be an integer register -- `mov r12,#0'
dvb/dvbsenco8.s:223: Error: operand 1 must be an integer register -- `str r12,[r1,#0]'
dvb/dvbsenco8.s:224: Error: operand 1 must be an integer register -- `str r12,[r1,#4]'
[... 54 lines cut ...]
dvb/dvbsenco8.s:342: Error: operand 1 must be an integer register -- `ldr r3,=interleavetable#:r3=base address of the interleave offset table'
dvb/dvbsenco8.s:343: Error: operand 1 must be an integer register -- `ldr r4,=packetcount#:r4=address of input packet number'
dvb/dvbsenco8.s:344: Error: operand 1 must be an integer register -- `ldr r5,[r4]#:r5=input packet count'
dvb/dvbsenco8.s:345: Error: operand 1 must be an integer or stack pointer register -- `and r5,#(NIPACKETS-1)#:mask off actual number of output packet slots'
dvb/dvbsenco8.s:346: Error: operand 1 must be an integer register -- `mov r6,#IPACKETSIZE'
dvb/dvbsenco8.s:347: Error: operand 1 must be a SIMD vector register -- `mul r5,r6#:r5=offset into ipackets of first byte to be filled'
dvb/dvbsenco8.s:348: Error: operand 1 must be an integer register -- `mov r6,#204'
dvb/dvbsenco8.s:350: Error: operand 1 must be an integer register -- `ldrh r7,[r3],#2#:get a 16 bit offset value from the interleave table and increment'
dvb/dvbsenco8.s:351: Error: operand 1 must be an integer register -- `ldrb r8,[r0],#1#:get an input byte'
dvb/dvbsenco8.s:352: Error: operand 1 must be an integer or stack pointer register -- `add r7,r5#:r7=offset into ipackets for this input byte'
dvb/dvbsenco8.s:353: Error: operand 1 must be an integer or stack pointer register -- `and r7,r2#:mask r7 for wrapping'
dvb/dvbsenco8.s:354: Error: operand 1 must be an integer register -- `strb r8,[r1,r7]#:store the input byte into ipackets'
dvb/dvbsenco8.s:355: Error: operand 1 must be an integer register -- `adds r6,#-1'
dvb/dvbsenco8.s:358: Error: operand 1 must be an integer register -- `ldr r1,=packetcount#:packet count'
dvb/dvbsenco8.s:359: Error: operand 1 must be an integer register -- `ldr r2,[r1]#:get the packet count'
dvb/dvbsenco8.s:360: Error: operand 1 must be an integer or stack pointer register -- `add r2,#1#:count a packet processed'
dvb/dvbsenco8.s:361: Error: operand 1 must be an integer register -- `str r2,[r1]#:save packet count'
dvb/dvbsenco8.s:363: Error: operand 1 must be an integer register -- `ldr r1,=ipackets'
dvb/dvbsenco8.s:364: Error: operand 1 must be an integer or stack pointer register -- `add r0,r1,r5#:return with output packet address'
dvb/dvbsenco8.s:366: Error: unknown mnemonic `pop' -- `pop {r1,r2,r3,r4,r5,r6,r7,r8}'
dvb/dvbsenco8.s:367: Error: unknown mnemonic `bx' -- `bx lr'
```

I'm stuck at this point.
Please include instructions on this so called beta mode please.

FalcoGer, 2020-08-03:
I managed to get it to compile by removing the `../dvbrf` target from all. Not sure what the implications are. Also not sure if it actually transmits. Have to wait for my SDR hardware to verrify

F5OEO, 2020-11-03:
Thanks for your advices, 64os is now supported.

closed 2020-11-03 by F5OEO