F5OEO/rpitx #216: Aparently crashes uart on pi 0 w..

M0NWY, 2020-08-24:
I'm attempting to use rpitx as the rf generator for an aprs tracker, but I loose coms with the gps chip after transmitting. Restarting gpsd doesn't fix it, power cycling gps chip doesn't fix it. reboot (warm - no powercycle) does fix. Smells like the uart on the chip is crashing. Not sure why. Am investigating.

F5OEO, 2020-08-24:
Never used the serial at same time of rpitx. As rpitx move some PLL the serial clock is maybe using it to derive its clock.

M0NWY, 2020-08-24:
That makes sense... I'll see what happens with a usb serial dongle.

M0NWY, 2020-08-24:
ok, doesn't crash a usb ftdi ( possibly fake ) dongle ! 
Dunno if you'd like to try and make it not crash the built in uart, but it'd be nice .
Cheerrs. xx.

M0NWY, 2020-08-24:
Hummmmm...... so it looks like although this doesn't actively crash a usb dongle, it does kill the data for about 3 seconds !