F5OEO/rpitx #218: Unable to determine hardware version

BlckhtX64, 2020-09-11:
Hi everyone,

I'm trying to transmit  with my Raspberry-Pi4B and SDR  dongle receiver.  I can receive frequencies but not transmit.

 I know that Rpi4 is in beta mode, but , It's is possible to transmit to 433Mhz frequency?

I run the next command:

rtl_sdr -s 250000 -g 19.2 -f 433.563.100 start.iq
Found 1 device(s):
  0:  Realtek, RTL2838UHIDIR, SN: 00000001

Using device 0: Generic RTL2832U OEM
Found Fitipower FC0012 tuner
Exact sample rate is: 250000.000414 Hz
Sampling at 250000 S/s.
Tuned to 433 Hz.
Tuner gain set to 19.20 dB.
Reading samples in async mode...
Allocating 15 zero-copy buffers

*******************************************************
Once I have caught the signal, then I run the rpitx command to transmit the same frequency  with:

******************************************************

rpitx -s 250000 -f 433.563.100  -i start.iq 

****
But:
****

Unable to determine hardware version. I see: Hardware : BCM2711
,
 - expecting BCM2708 or BCM2709. Please report this to projects@drogon.net
rpitx Version 0.1 compiled Sep 11 2020 (F5OEO Evariste) running on  Frequency=0.000000  SampleRate=250000 

*************************************************************************************
In addition, When I try with the sendiq comand, the output is the Terminate method caught.
*************************************************************************************

"Terminate, Signal caught 8"

Anyone have had this problem too?

Thank you.

F5OEO, 2020-09-11:
Seems Librpitx in beta for rpi4 was not really tested..I will merged it with pi4 beta soon. You will have version to your pi4.

BlckhtX64, 2020-09-11:
Thank you for your response. I would like help you to develop the functions that have not implemented yet,but is hard for the first time if I don't know the code. I don't like only post the question and that's all.

I have Raspberry pi 4 with Broadcom chip BCM2711. Quad core 64 bits ARM Cortex A72 up to 1,5Ghz .

For now it seems that only works for BCM2837 or Raspberry pi 3+ and older.

I would like do my research about HackingRF with my own transceiver and not via HackRF that give all done.

I hope you will fix this soon . Thank you!!

F5OEO, 2020-11-03:
Should be fixed with current commit.

closed 2020-11-03 by F5OEO