F5OEO/rpitx #219: Please add POCSAG  GPIO output for external transmitter.

lolren, 2020-09-12:
Rpi-tx pocsag is great, but it`s messy , even with filter. If we had a dedicated pin we can setup for output the POCSAG signal , we can connect a module like a HopeRf module to call our pagers. Thank you

VasylSamoilov, 2020-09-22:
Maybe you can utilize pocsag implementation for arduino that interface with the si4432 based ISM-tranceiver for broadcasting the message? See https://github.com/on1arf/rf22_pocsag . There are a lot of POCSAG implementations out there.

catSIXe, 2021-02-23:
> Rpi-tx pocsag is great, but it`s messy , even with filter. If we had a dedicated pin we can setup for output the POCSAG signal , we can connect a module like a HopeRf module to call our pagers. Thank you

Take a look at the RFM69 its a hardware FSK modem and you can send POCSAG messages with it. 
Im currently implementing a Duplex RX/TX RFM69 POCSAG Gateway in Golang.