F5OEO/rpitx #220: Tools do not build and install properly

matthiasseemoo, 2020-09-13:
There seem to be two problems in the Makefile:
1. sendook does not build as it does not find atouint32_metric.
2. install does not work, as foxhunt is not built by default.

The following change to the Makefile fixes the issues:

```
diff --git a/src/Makefile b/src/Makefile
index 994b388..06c2da1 100644
--- a/src/Makefile
+++ b/src/Makefile
@@ -1,4 +1,4 @@
-all: ../pisstv ../piopera ../pifsq ../pichirp ../sendiq ../tune ../freedv ../dvbrf ../pocsag ../spectrumpaint ../pifmrds ../rpitx ../corel8 ../pift8 ../sendook ../morse
+all: ../pisstv ../piopera ../pifsq ../pichirp ../sendiq ../tune ../foxhunt ../freedv ../dvbrf ../pocsag ../spectrumpaint ../pifmrds ../rpitx ../corel8 ../pift8 ../sendook ../morse
 
 CFLAGS = -Wall -g -O2 -Wno-unused-variable
 CXXFLAGS = -std=c++11 -Wall -g -O2 -Wno-unused-variable
@@ -69,8 +69,8 @@ LDFLAGS_Pissb = librpitx/src/librpitx.a -lm -lrt -lpthread -lsndfile -lliquid -L
 ../pift8 : pift8/pift8.cpp pift8/ft8_lib/ft8/constants.cpp pift8/ft8_lib/ft8/text.cpp pift8/ft8_lib/ft8/pack.cpp pift8/ft8_lib/ft8/encode.cpp librpitx/src/librpitx.a
        $(CCP) $(CXXFLAGS) -Wno-write-strings -o ../pift8 pift8/pift8.cpp pift8/ft8_lib/ft8/constants.cpp pift8/ft8_lib/ft8/text.cpp pift8/ft8_lib/ft8/pack.cpp pift8/ft8_lib/ft8/encode.cpp  $(LDFLAGS)
 
-../sendook: ook/sendook.cpp librpitx/src/librpitx.a
-       $(CCP) $(CXXFLAGS) -Wno-write-strings -o ../sendook ook/sendook.cpp $(LDFLAGS)
+../sendook: ook/sendook.cpp librpitx/src/optparse.c librpitx/src/librpitx.a
+       $(CCP) $(CXXFLAGS) -Wno-write-strings -o ../sendook ook/sendook.cpp librpitx/src/optparse.c $(LDFLAGS)
 
 CFLAGS_Pifm    = -Wall -g -O2 -Wno-unused-variable
 LDFLAGS_Pifm   = librpitx/src/librpitx.a -lm -lrt -lpthread -lsndfile -L/opt/vc/lib -lbcm_host
```

lolren, 2020-09-13:
just noticed that as well..thank you for the patch, sadly, it does not work for me:(

F5OEO, 2020-09-16:
Thanks for the patch @matthiasseemoo .

closed 2020-09-16 by F5OEO