F5OEO/rpitx #223: Making rpitx work with arm64/Kali Pi

FawkesOficial, 2020-09-19:
I decided to install rpitx on my raspberry pi 3b+ that is running kali linux:
`root@kali:~# uname -a`
`Linux kali 4.19.118-Re4son-v8+ #1 SMP PREEMPT Wed May 6 12:44:21 UTC 2020 aarch64 GNU/Linux `

Turns out that the installl process wasn't that plug&play.

The reason why I'm opening this issue is to help others who may have faced similar issues while getting rpitx to work.
Here's a list of things I did and edited to make it work (So far it has worked fine).

**_Also, I want to mention that I also ran this:_**
`apt install g++-arm-linux-gnueabihf gcc-arm-linux-gnueabihf`
`dpkg --add-architecture armhf`
`apt-get install -y libfftw3-dev:armhf`
**_But I have no idea if it afected anything_**

### This was how I installed rpitx:
`git clone https://github.com/F5OEO/rpitx `
`cd rptx/`
`nano install.sh`

**Inside "install.sh" I commented the line** `git clone https://github.com/F5OEO/librpitx` **and saved the file.**

`cd src/`
`nano Makefile`

**Inside "Makefile" I commented:**
`../dvbrf : dvb/dvbrf.cpp dvb/dvbsenco8.s dvb/fec100.c dvb/dvbs2arm_1v30.s librpitx/src/librpitx.a`
`       $(CC) $(CFLAGS) -c -o dvb/dvbsenco8.o dvb/dvbsenco8.s`
`       $(CC) $(CFLAGS) -c -o dvb/dvbs2arm_1v30.o dvb/dvbs2arm_1v30.s`
`       $(CC) $(CFLAGS) -c -o dvb/fec100.o dvb/fec100.c`
`       $(CCP) $(CXXFLAGS) -o ../dvbrf dvb/dvbrf.cpp dvb/dvbsenco8.o dvb/fec100.o dvb/dvbs2arm_1v30.o $(LDFLAGS)`

**and the line** `       install -m 0755 ../dvbrf /usr/bin` **and saved the file.**

**Then I manually clonned "librpitx"** `git clone https://github.com/F5OEO/librpitx`
`cd librpitx/src/`
`nano atv.cpp`

**Inside "atv.cpp" I changed the line:**
`    dbg_printf(1, "Last cbp :  %d \n", ((unsigned int)(cbp) - (unsigned int)(cbarray)) / sizeof(dma_cb_t));`
**to:**
`    dbg_printf(1, "Last cbp :  %d \n", ((uintptr_t)(cbp) - (uintptr_t)(cbarray)) / sizeof(dma_cb_t));`

**Finally I went back to rpitx main folder and ran:** `./install.sh`

### That's it
Hopefully I can help someone with this.
I would also like to link some sources to the fixes that I presented here:

- https://github.com/F5OEO/librpitx/pull/13/commits/d765afbf2aba7fa76d952717f01636f2f111a095
- https://github.com/F5OEO/rpitx/issues/215#issuecomment-668193843
- https://github.com/F5OEO/rpitx/issues/173
- https://github.com/F5OEO/librpitx/issues/11

Also, excuse me if this issue isn't really well-written but it's my first :)

F5OEO, 2020-09-19:
Thanks for pointing issues with 64bits..Sorry never used 64bit os on raspberry right now. I will commit based on your comments.

F5OEO, 2020-11-03:
Thanks for your big work...Compilation is ok now on Raspbian OS64 so should work on Kali64.

closed 2020-11-03 by F5OEO

FawkesOficial, 2020-11-03:
> Thanks for your big work...Compilation is ok now on Raspbian OS64 so should work on Kali64.

Np, always here to help ;)