F5OEO/rpitx #227: Strange timing problems

clemibunge, 2020-11-02:
Well, the problem is that everything is sped up -> example video( https://imgur.com/a/SrILgKu ) even POCSAG and opera and SSTV so it doesn't seem to be related to any sample rate issues but rather to some timing library I don't know... I guess some library that is installed via APT changed something...
So I just installed Rpitx on my RPI 3B and it had the described timing issues tried a second board cuz I thought it was faulty.. it wasn't ... Tried a raspberry pi zero with had the same issues... I did a clean install of Raspian stretch every time and installed nothing else just rpitx

F5OEO, 2020-11-03:
I am working on release the pi4 version, but did a wrong testing before commit to master ! The version is unstable at the moment.

F5OEO, 2020-11-03:
Should be fixed ! Can you try to reinstall to confirm it is working at proper speed ?

clemibunge, 2020-11-03:
I'll test it in two hours still in school right now

clemibunge, 2020-11-03:
the timing works now! but the frequency is kinda inaccurate by ~5Khz

F5OEO, 2020-11-03:
Have your raspberry an internet connexion (for ntp)?

clemibunge, 2020-11-03:
I do have an internet connection for the RPI's yes
I tried transmitting on 446.006Mhz with the pi zero and the frequency was off by -8Khz
on the pi 3, the frequency is off by about +2Khz  I guess that's normal?

clemibunge, 2020-11-03:
ah and btw u disabled dvbrf right? you forgot to remove it from the makefile i got an not found error while installing

F5OEO, 2020-11-03:
Sure ? https://github.com/F5OEO/rpitx/commit/d1d8a37f95b5f061f3656dddb80e7cfd0a1916ff

clemibunge, 2020-11-03:
yes, sure look at line 101

clemibunge, 2020-11-03:
frequency got more stable now(~0.5khz off) NTP did its job 

closing the issue

closed 2020-11-03 by clemibunge