F5OEO/rpitx #231: RPITX with Pi4 beta - Monitor going blank - cannot recover

pmmeasures-tech, 2020-11-08:
RPITX with Pi4 beta - Monitor going blank when mode in operation
Cannot Ctrl+C back to menu monitor says refresh rate unsupported
only option is reboot - system at suggest GPU speed 250Mhz
But is working out visible on SDR just something funky around HDMI monitor.

F5OEO, 2020-11-08:
Maybe some resolution could not be achieved with HDMI. Rpitx is mainly used in headless (no hdmi monitor). Need to further investigate.

pmmeasures-tech, 2020-11-08:
Appears that way, was running 1920x1080 and it failing,after reboot dropped to a low res and it worked ok went back up to 1280x1024 and it failed. so a funny there on screen implementation when it drops into a main function. I get the initial menu ok and e.g. if I selected "Chirp" at that point screen goes blank and does not recover but the function does run and I can see output on SDR.  will check later in week what resolutions do work.

VK3ZTE, 2020-11-17:
Am experiencing same issue with 1920x1080.  Reducing resolution down to 620x480 before launching rpitx solves (sort of) the problem.

Ben1701A, 2020-12-26:
The Pi 4 is still in beta...Maybe rpitx should suggest a lower resolution than 1280x1024 during install when the model is a match?

Ben1701A, 2021-01-08:
Also seeing similar behavior on the Pi 400. I was using a tablet via VNC as a monitor at that resolution, and nothing happened when I ran the TX. However, I connected it to a 1440x900 monitor and it crashes the video output, even if it were the same exact resolution. It doesn't crash the Pi itself though, and I was able to blindly type `init 6` into that terminal window I had open and reboot.

Ben1701A, 2021-01-16:
Huh. ChristopheJacquet/PiFmRds still works without issues. Anyone still on this issue?

MotaDan, 2021-01-26:
Another data point, I'm on Ubuntu mate and going to 640 x 480 did not help. Also I've overclocked my rpi4 8GB to arm_freq=2147
and gpu_freq=750. I commented out the part of the install script that edited the config file. Since the config file is in a different place and I didn't want to drop the gpu frequency or have it running at max speed all the time.

SnkMtn000, 2021-02-14:
I run headless, VNC and/or ssh and still get the issue periodically.  I also use heat sinks and fans and monitor temp and cpu usage.  When I have no issues I see nothing to indicate either a cpu or temperature problem, but during a fail, wifi disconnects and transmission continues as monitored on a radio receiver.  Odd.

1)  Does anyone know of a way to measure the transmitter impedance at gpio4 pin 7, used for the antenna?  What if not 50 ohms?  Could power reflected back into the pi be at issue with stability?

2)  Read a good hackaday article about the pi's power supply issues causing kernel panics.  I recall those with early rpi v1.0 and cheap wall warts until I upgraded to a higher current wart.  The article also mentioned piggy backing a massive 470 uF cap across the 5v line.

Thoughts.

Charles

hydrogen18, 2021-11-16:
> I run headless, VNC and/or ssh and still get the issue periodically. I also use heat sinks and fans and monitor temp and cpu usage. When I have no issues I see nothing to indicate either a cpu or [...]

1. The reflected power couldn't exceed the peak output power, which is about 10 milliwatts or so. That isn't going to cause a problem.
2. Yes, crap power supplies cause issues with almost any board. Capacitors can help with the problem but using better power supplies is also a good idea.

Jibun-no-Kage, 2022-02-19:
Running on Pi4 with very good supply, using UART serial oonsole... the output to screen is random garbage, and transmission seems to fail,  In fact, the console seems to crash, have to reboot to recover function.

Running the same command on the same system on ssh session via PuTTY, the same command, sendiq runs shows the expected output, see below, but transmission seems to fail.

Output in PuTTY session as note above...
`# sendiq -s 250000 -f 303.87500e6 -t u8 -i /usr/local/bin/rpitx/signals/Casitas/Light.iq
Peri Base = fe000000 SDRAM c0000000
RPi4 GPIO detected
Peri Base = fe000000 SDRAM c0000000
RPi4 GPIO detected
Peri Base = fe000000 SDRAM c0000000
RPi4 GPIO detected
Peri Base = fe000000 SDRAM c0000000
RPi4 GPIO detected
Peri Base = fe000000 SDRAM c0000000
RPi4 GPIO detected
Peri Base = fe000000 SDRAM c0000000
RPi4 GPIO detected
Peri Base = fe000000 SDRAM c0000000
RPi4 GPIO detected
Peri Base = fe000000 SDRAM c0000000
RPi4 GPIO detected
Peri Base = fe000000 SDRAM c0000000
RPi4 GPIO detected
Peri Base = fe000000 SDRAM c0000000
RPi4 GPIO detected
Peri Base = fe000000 SDRAM c0000000
RPi4 GPIO detected
Peri Base = fe000000 SDRAM c0000000
RPi4 GPIO detected
Peri Base = fe000000 SDRAM c0000000
RPi4 GPIO detected
End of file
`

The same installation, i.e. SD card on a Model 2B and Model 3B works without issue.

hydrogen18, 2022-02-20:
> The same installation, i.e. SD card on a Model 2B and Model 3B works without issue. 

What board did you compile on?

Jibun-no-Kage, 2022-02-20:
For the 2B and 3B I compiled on the 3B.  Did not seem to need to recompile on the 2B, just copied files.  These worked fine.  For the 4B, I tried the 3B compiled files, did not work, so did fresh compile on 4B, but still did not work.  Plan to do some more testing later today when I get a chance.

Jibun-no-Kage, 2022-03-13:
Did some more testing, and I have now seen serial console 'corruption' if you will, on a Pi Model 3B, not just 4B.