F5OEO/rpitx #232: Ctrl C not working and intermittent black band

Murphyv310, 2020-11-12:
Hi.
I have installed this on my Pi4b 4gb. it is installed on Raspberry Pi OS lite the latest release.  The install went fine and I updated the config by agreeing, typing Y. I can get to the frequency and changing it on pressing enter it goes to the menu as it should but when I use a command in the menu I cannot return back to the menu by pressing Control C. So I have to reboot if I want to do something else. Also on around 60% of the time I get a big black band on the screen. I've tried this on another Pi4 2gb with the same result. 
Can anyone give me a simple answer? 
Also how do you insert audio etc, is the only way to access this program through easytest? I'm not into code or programming unfortunately.

F5OEO, 2020-11-12:
Rpitx and HDMI output is not yet really tested. Can you try without HDMI output (through network ssh) to see if it solves the issue.

Murphyv310, 2020-11-12:
Hi.
Thanks for the reply. How do I do that?

Murphyv310, 2020-11-15:
Hi.
Can anyone help before I give up on this project that I've been patiently waiting for, for months on end.