F5OEO/rpitx #234: Missing libbcm_host.so on 64 bit Manjaro

themoonwatcher, 2020-11-25:
On Manjaro linux there is no /opt/vc/lib where the compiler is looking for bcm_host.so. What can be done to make it run?

F5OEO, 2020-11-25:
You maybe miss some files in https://github.com/raspberrypi/firmware

themoonwatcher, 2020-11-25:
Sure, the /opt directory is empty. I guess Manjaro does need them. I've tried copying libbcm_host.so from the link you give above, but then the compiler says "/usr/bin/ld: skipping incompatible /opt/vc/lib/libbcm_host.so".

F5OEO, 2020-11-25:
You need to get the 64bit version...

themoonwatcher, 2020-11-29:
I've managed to find the source for Raspberry's userland-arm64 and found the needed compilers in /usr/bin/aarch-* on Manjaro's distribution.
I've went manually trough install script and compiled and installed everything, including the /boot/config.txt changes.
When I run easytest.sh nothing is being transmitted.
Where should I start debugging?
What may be wrong?
Do you want some more info to help you pin the problem?

By the way the easytest script uses unnecessary space to display and is being cut off on the Rpi's official display.