F5OEO/rpitx #235: make chirp sweep in one direction

giordyb, 2020-12-05:
Hi I'd like to use rpitx as a frequency sweeper in order to test some filters with an oscilloscope.
Is there a way to modify chirp so that it sweeps in one direction only? 
I tried to look at the code but I cannot figure out what is being passed to the SetFrequencySample() function, c is not my first language...

F5OEO, 2020-12-06:
See https://github.com/F5OEO/rpitx/blob/7339faad4b2178a92a912c0a1cc09da8b28e7af1/src/chirp/chirp.cpp#L94

giordyb, 2020-12-06:
Thank you!

I was able to do something close to what I was looking for by modifying the tune.cpp and doing a loop calling the SetFrequency function.

Would you mind explaining the difference between the approach of the tune.cpp and chirp.cpp?
It looks like they both take a frequency but it seems that chirp is a lot faster switching frequency.