F5OEO/rpitx #240: What PWM frequency is used

criede, 2021-01-06:
Hello Evariste,
two Questions:
1. Is the PWM frequency dependant on the output frequency or is always the same PWM frequency used for all output frequencies?
2. What exactly is the PWM frequency, respectively how is it calculated?
Unfortunately there are only few comments in the code, so getting the answer form the code is not a quick thing for me, so maybe you can shorthen the path for me?
Thanks and 73 de Christian DB9CR

F5OEO, 2021-01-06:
Pwm is used in patern mode...It is based on the PLL frequency (like the clock output).
Instead of just output a divisor of pll frequency in CLK mode, we use a 32 bit patern which will be output.
This is very convenient to make some phase modulation...see https://github.com/F5OEO/librpitx/blob/master/src/phasedmasync.cpp
If you want 4 phase, just change patern at the samplerate : 
Phase 1: 0011, Phase 2 0110, Phase 3 1100 , Phase 4 1001 
Hope that it could help you understanding pwm mode. Let me know.

criede, 2021-01-08:
Thanks for the explanation. I meanwhile found  https://groups.io/g/rpitx/topic/28976987#49 that also answers some of my questions.
Can I encourage you to write some more high level documentation about how all this works, maybe in the readme? Also, some comments in the code would increase readability a lot.