F5OEO/rpitx #244: RPi 4 crashes on

simpson-peter, 2021-01-14:
Whenever I run any of the frequencies on ./easytest.sh my RPi4 stops all visual output to my monitor, and I seemingly can't control the rpi with any peripherals (ctrl + c or ctrl + d doesn't fix the problem). The signal however does play so the rpi is still functional. Has anyone had this problem? Is it a power problem, or just an rpi4 beta issue?

Ben1701A, 2021-01-16:
It definitely happens with direct monitor output. When I was running a VNC server without HDMI output, I could run high resolutions easily while running rpitx. It seems like it's a rpi4 beta issue.

Also, this issue was already created.

Jibun-no-Kage, 2022-02-18:
Can you connect to the serial port, i.e. UART?

Jibun-no-Kage, 2022-02-19:
Tested Pi4 beta, corrupted serial console output and freezes serial console when trying to run sendiq for example.  Running from PuTTY session I get the right output, but the transmission fails it appears.  Tried the same command on Model 2B and 3B and it works fine.

Jibun-no-Kage, 2022-03-13:
DId some more testing,and now have seen serial port 'corruption' on a 3B, similar to the 4B serial port issues.  The difference is, on the 3B (via PuTTY) session, RPiTx works as expected, whereas on 4B I still can't seem to get it to work at all.  Have tired 4 different Pi 4B devices.  No joy so far.

Ben1701A, 2022-03-20:
Aight.
So I got the `tune` binary working on the Pi 400, but anything else (tried chirp) seems to kill video, as well as the network interface (unable to SSH or ping).
`tune` also seems to take my inputted frequency and double it.
```
$ sudo ./tune -f 140000000
Peri Base = fe000000 SDRAM c0000000
RPi4 GPIO detected
Running on Pi4
```
(repeated 4 more times...)
```
Info:NTP find offset 130778 freq 373840 pps=0 ppm=5.704346
Pi4=1 Xosc = 54000000 Freq PLL no 5= 647996303
Found PLL solution for frequency 280.0MHz : divider:5 VCO: 1400.0MHz
Master PLLC locked
```
I'm running this on Ubuntu 21.10 arm64.

Ben1701A, 2022-03-20:
o>0
ohh wait a second...
I did the following and the screen momentarily blanked. Maybe we're getting a completely blank screen from the programs that have change in their signal constantly (i.e. pifmrds, or anything else in easytest other than tune).
```
sudo ./tune -f 1400000000
sudo ./tune -f 700000000
```
The top one reports that it's too high for `tune` to reach.
The bottom command continues running _after_ the monitor blank.
There's also some change in what the bottom command reports:
```
Pi4=1 Xosc = 54000000 Freq PLL no 5= 11199999988
Found PLL solution for frequency 1400.0MHz : divider:1 VCO: 2800.0MHz
Pll Fix Divider
Use PLL Prediv
Master PLLC Locked
```
`Fix Divider'. >.>

pschonmann, 2022-06-26:
Hi ! Same shit here !
Im running debian 11 on pi4 with pimox7, just only over ssh, no hdmi into monitor.

./easytest.sh - works fine except commands using sendiq - this crash my rpi

Got output below and PI restarted.

```
5.15.32-v8+ #1538 SMP PREEMPT Thu Mar 31 19:40:39 BST 2022 aarch64 GNU/Linux

cat /etc/debian_version 
11.3
```

```
root@pimox01:~/petr/rpitx# ./testssb.sh 
csdr bandpass_fir_fft_cc: window = HAMMING
csdr bandpass_fir_fft_cc: (fft_size = 1024) = (taps_length = 401) + (input_size = 624) - 1
(overlap_length = 400) = taps_length - 1
csdr fir_interpolate_cc: window = HAMMING
csdr fir_interpolate_cc: taps_length = 79
Peri Base = fe000000 SDRAM c0000000
RPi4 GPIO detected
Peri Base = fe000000 SDRAM c0000000
RPi4 GPIO detected
Peri Base = fe000000 SDRAM c0000000
RPi4 GPIO detected
Peri Base = fe000000 SDRAM c0000000
RPi4 GPIO detected
Peri Base = fe000000 SDRAM c0000000
RPi4 GPIO detected
Peri Base = fe000000 SDRAM c0000000
RPi4 GPIO detected
Peri Base = fe000000 SDRAM c0000000
RPi4 GPIO detected
Peri Base = fe000000 SDRAM c0000000
RPi4 GPIO detected
Peri Base = fe000000 SDRAM c0000000
RPi4 GPIO detected
Peri Base = fe000000 SDRAM c0000000
RPi4 GPIO detected
Peri Base = fe000000 SDRAM c0000000
RPi4 GPIO detected
Peri Base = fe000000 SDRAM c0000000
RPi4 GPIO detected
Peri Base = fe000000 SDRAM c0000000
RPi4 GPIO detected
client_loop: send disconnect: Broken pipe

or 

root@pimox01:~/petr/rpitx# rpitx -m IQ -i 433B.iq  -f 433.940
Warning : rpitx V2 is only to try to be compatible with version 1
Peri Base = fe000000 SDRAM c0000000
RPi4 GPIO detected
Peri Base = fe000000 SDRAM c0000000
RPi4 GPIO detected
Peri Base = fe000000 SDRAM c0000000
RPi4 GPIO detected
Peri Base = fe000000 SDRAM c0000000
RPi4 GPIO detected
Peri Base = fe000000 SDRAM c0000000
RPi4 GPIO detected
Peri Base = fe000000 SDRAM c0000000
RPi4 GPIO detected
Peri Base = fe000000 SDRAM c0000000
RPi4 GPIO detected
Peri Base = fe000000 SDRAM c0000000
RPi4 GPIO detected
Peri Base = fe000000 SDRAM c0000000
RPi4 GPIO detected
Peri Base = fe000000 SDRAM c0000000
RPi4 GPIO detected
Peri Base = fe000000 SDRAM c0000000
RPi4 GPIO detected
Peri Base = fe000000 SDRAM c0000000
RPi4 GPIO detected
Peri Base = fe000000 SDRAM c0000000
RPi4 GPIO detected
client_loop: send disconnect: Broken pipe
```

andywei2015, 2023-11-11:
the same problemwhen running rpitx easytest command with raspberry 4b. hdmi signal lost. any solutions?

comicsansgreenki, 2023-11-11:
@andywei2015 to my knowledge, there's only two ways around, each with their own tradeoff:

- Continue running an HDMI output, but you're limited to an uncertain range of frequencies on `tune` with some trial and error. PiFmRds runs just fine though!
- Or _don't_ use the HDMI output. This'll let you use the whole suite of commands without issue. 
 - **This doesn't prevent you from using the VNC Server to stream the monitor instead!** If you choose to go this way, the whole thing is done through `raspi-config` - just set your preferred screen resolution, enable VNC server, and reboot.

andywei2015, 2023-11-11:
> @andywei2015 to my knowledge, there's only two ways around, each with their own tradeoff:
> 
> * Continue running an HDMI output, but you're limited to an uncertain range of frequencies on `tune` with some trial and error. PiFmRds runs just fine though!
> * Or _don't_ use the HDMI output. This will let you use the whole suite of commands without issue.
>   
>   * **This doesn't prevent you from using the VNC Server to stream the monitor instead!** If you choose to go this way, the whole thing is done through `raspi-config` - just set your preferred screen resolution, enable VNC server, and reboot.

many thanks for your kind reply.