F5OEO/rpitx #247: Compilation does not work (librpitx.h no such file or directory)

catSIXe, 2021-02-23:
1. Fresh install of latest lite raspberrypios
2. `sudo apt-get update`
3. `sudo apt-get install git`
3. `git clone https://github.com/F5OEO/rpitx`
4. `cd rpitx`
5. `./install.sh`
```
install libft8.a /usr/lib/libft8.a
g++ -std=c++14 -I.   -c -o pift8.o pift8.cpp
pift8.cpp:13:10: fatal error: librpitx/librpitx.h: No such file or directory
 #include <librpitx/librpitx.h>
          ^~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make: *** [<builtin>: pift8.o] Error 1
g++ -std=c++11 -Wall -g -O2 -Wno-unused-variable -o ../pisstv sstv/pisstv.cpp  -L/opt/vc/lib -lrpitx -lm -lrt -lpthread
sstv/pisstv.cpp:18:10: fatal error: librpitx/librpitx.h: No such file or directory
 #include <librpitx/librpitx.h>
          ^~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make: *** [Makefile:17: ../pisstv] Error 1
g++ -std=c++11 -Wall -g -O2 -Wno-unused-variable -o ../pisstv sstv/pisstv.cpp  -L/opt/vc/lib -lrpitx -lm -lrt -lpthread
sstv/pisstv.cpp:18:10: fatal error: librpitx/librpitx.h: No such file or directory
 #include <librpitx/librpitx.h>
          ^~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make: *** [Makefile:17: ../pisstv] Error 1
```

Complete Log : https://pastebin.com/7Dp1xjPD

the librpitx repository gets cloned and compiled

Jaki0571, 2021-02-25:
same problem! Can your Pi transmit signals normally?  My Pi 4B can't transmit because ./pifmrds doesn't generate

catSIXe, 2021-02-25:
> <author@noreply.github.com>
> such file or directory) (#247)

I have compiled it on a RPI Zero W , some time ago. i just copied the 
binaries and it worked on a 3B
------ Origineel bericht ------

catSIXe, 2021-02-25:
I have compiled it on a RPI Zero W , some time ago. i just copied the
binaries and it worked on a 3B

Jaki0571, 2021-02-25:
I tried to copy the FMRDS execution files from other projects and it did work, but I don't know if it was the new RaspberryPIOS because I was compiling on the old PIOS a few days ago and there was no problem

F5OEO, 2021-02-25:
Should be fixed with https://github.com/F5OEO/librpitx/commit/e0a4588df8b4ff5d6008b2fb3e2a193c239b8202

Jaki0571, 2021-02-25:
GOOD JOB! it's working! thank you very much!

closed 2021-03-01 by catSIXe

petitxef, 2023-12-12:
Just an update, it's show the same problem: https://github.com/F5OEO/rpitx/issues/311