F5OEO/rpitx #248: sendiq and rpitx not doing anything on a pi3

sebastianheyn, 2021-03-04:
Hey guys,

I want to send an IQ file but I cannot get sendiq or rpitx to send anything. The iq file has been created using sdrsharp and has a samplerate of 312.5khz

now when I try 
./tune 99.9e6 -> it works
./easytest.sh -> works

./sendiq -f 99.9e6 -i 10-04-27_103600kHz.wav -s 312.5e3  -p 7 -d
Peri Base = 3f000000 SDRAM c0000000
Peri Base = 3f000000 SDRAM c0000000
Peri Base = 3f000000 SDRAM c0000000
Peri Base = 3f000000 SDRAM c0000000
Peri Base = 3f000000 SDRAM c0000000
Peri Base = 3f000000 SDRAM c0000000
Peri Base = 3f000000 SDRAM c0000000
Peri Base = 3f000000 SDRAM c0000000
Peri Base = 3f000000 SDRAM c0000000
Peri Base = 3f000000 SDRAM c0000000
Peri Base = 3f000000 SDRAM c0000000
Peri Base = 3f000000 SDRAM c0000000
Peri Base = 3f000000 SDRAM c0000000

but not even a carrier is beeing output. Any idea??

Ghaz75, 2021-03-22:
> Hey guys,
>
> I want to send an IQ file but I cannot get sendiq or rpitx to send anything. The iq file has been created using sdrsharp and has a samplerate of 312.5khz
>
> now when I try
> ./tune 99.9e6 -> [...]

The main culprit is the parameter force_turbo=1 with default 1. I had this value in /boot/config.txt along with gpu_freq=250. In my Raspbian Stretch With Desktop OS I made sure that there is no over clocking. I left the gpu_freq=250 and changed the force_turbo=1 to 0 and gave a reboot. And Voila.
But for AM mode the voice is like chipmunks. Rest all modes are working fine.
For Raspbian Buster try force_turbo=0, so far I have no success on raspbian buster.
Thanks.

Jibun-no-Kage, 2022-02-18:
Just tested on Model 2B with turbo=1 in place, was able to do sendiq without issue, controlling a ceiling fan at 303.7500e6 frequency, rate 20000, type u8, power default level.